rogerclarkmelbourne/Arduino_STM32 #694: sdiof1 library example

serkanc, 2019-09-27:
Hi,

I need sample sdiof1 library example to try sdio port on stm32f103ret breakout board.
I can't find any example for the library.
Thanks

stevstrong, 2019-09-27:
You may adapt this [example for F4](https://github.com/stevstrong/SdFat/blob/master/examples/STM32F4SdioDemo/STM32F4SdioDemo.ino) to F1 (reduce buffer size).

serkanc, 2019-09-27:
There is an error when compiling
error: 'SdFatSdio' does not name a type
SdFatSdio sd;

I'm using Sdfat library from Bill Greiman v1.1.0

stevstrong, 2019-09-27:
You have to enable USE_SDIO in configuration file of SdFat.
Or use my SdFat fork.

stevstrong, 2019-09-27:
Insert
`#define ENABLE_SDIO_CLASS 1`
here:
https://github.com/greiman/SdFat/blob/7417ee943ccebbaf27f51042e2b3ae41e9defa45/src/SdFatConfig.h#L213

serkanc, 2019-09-27:
I have enabled USE_SDIO in configuration and added #define ENABLE_SDIO_CLASS 1, but still same error.

stevstrong, 2019-09-27:
Do you use Arduino IDE?
Alternatively, try other SDIO example from Greiman' repo.

serkanc, 2019-09-27:
I'm using Arduino IDE.
I tried example for TeensySDIO example, but still same error. I downloaded Sdfat Beta, same error..  
I will tell this issue. I will use your sdfat fork. 
How can I change buswidth to 1bit with this library... 
changing this line will be sufficent ?
  sdio_set_dbus_width(SDIO_CLKCR_WIDBUS_4BIT);

stevstrong, 2019-09-27:
I don't know.
Victor ported the SDIO lib from F4 (which only supports 4bit bus width) to F1 and made some modifications on that.

stevstrong, 2019-09-27:
You must be missing some installation detail if SDIo is not recognized.

serkanc, 2019-09-27:
https://github.com/victorpv/SdFat
I used this Sdfat... The same error...

stevstrong, 2019-09-27:
That fork of Victor is very old.
You should use the original one or my one.

serkanc, 2019-09-27:
https://github.com/stevstrong/SdFat
Still same error examples STM32F4SdioDemo, only changing SdioF4.h to SdioF1.h

serkanc, 2019-09-27:
C:\Users\serka\Documents\Arduino\libraries... I can use sdfat with normal SPI examples. But it give an error when I want to compile sdio examples

serkanc, 2019-09-27:
I'm using Generic stm32103R as board. variant Stm32103RE

stevstrong, 2019-09-27:
Try to compile for Generic STM32F103C (blue pill).
Double check the SdFat config file and be sure that the right headers are included in the sketch.

serkanc, 2019-09-27:
SdSpiSTM32.cpp: In member function 'uint8_t SdSpiAltDriver::receive(uint8_t*, size_t)':
error: invalid conversion from 'uint8_t {aka unsigned char}' to 'const void*' [-fpermissive]
   m_spi->dmaTransfer((uint8_t)0xFF, buf, n);
Example in Sdfat by stevstrong STM32test.... Sdfat Bill Greiman  works perfect with stm32f103 with spi ( SPIClass and Arduino SPi library, both works)
I

serkanc, 2019-09-27:
Using library SPI at version 1.0 in folder: C:\Users\serka\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2019.9.24\libraries\SP

Using library SdFat at version 1.0.14 in folder: C:\Users\serka\Documents\Arduino\libraries\SdFat

serkanc, 2019-09-27:
I passed "error: 'SdFatSdio' does not name a type" this writing  #define ENABLE_SDIO_CLASS 1 under #if defined(__STM32F1__) in SdFatConfig.h
but the error is 

Documents\Arduino\libraries\SdFat\src\SpiDriver\SdSpiSTM32.cpp:74:43: error: invalid conversion from 'uint8_t {aka unsigned char}' to 'const void*' [-fpermissive]
	m_spi->dmaTransfer((uint8_t)0xFF, buf, n);

In file included from C:\Users\serka\Documents\Arduino\libraries\SdFat\src\SpiDriver\SdSpiDriver.h:32:0,

                 from C:\Users\serka\Documents\Arduino\libraries\SdFat\src\SpiDriver\SdSpiSTM32.cpp:26:

C:\Users\serka\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2019.9.24\libraries\SPI\src/SPI.h:299:11: error:   initializing argument 1 of 'uint8 SPIClass::dmaTransfer(const void*, void*, uint16)' [-fpermissive]

     uint8 dmaTransfer(const void * transmitBuf, void * receiveBuf, uint16 length);

           ^

Using library SDIO in folder: C:\Users\serka\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2019.9.24\libraries\SDIO (legacy)
Using library SdFat at version 1.0.14 in folder: C:\Users\serka\Documents\Arduino\libraries\SdFat 
Using library SPI at version 1.0 in folder: C:\Users\serka\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2019.9.24\libraries\SPI 
exit status 1
Error compiling for board Generic STM32F103R series.

stevstrong, 2019-09-27:
This is because the core you are using does not have that specific SPI function.
I recommend you to take SPI lib from my repo. Or even better, take my complete repo instead of what you are using at the moment.

serkanc, 2019-09-27:
2019.9.24 stmduino STM32F1 ... what is your board manager url ?
I thought I''m using your repo ?

stevstrong, 2019-09-27:
Installation procedure: https://github.com/rogerclarkmelbourne/Arduino_STM32/wiki/Installation

For the ZIP file containing the STM32 files download use this link: https://github.com/stevstrong/Arduino_STM32/archive/master.zip

serkanc, 2019-09-27:
I copied your repo, deleted other stm2duino.com repo.... COmpiled for STM32f103R
C:\Users\serka\Documents\Arduino\hardware\Arduino_STM32\STM32F1\variants\generic_stm32f103r\wirish\boards.cpp:84:6: error: ambiguating new declaration of 'bool boardUsesPin(uint8)'
 bool boardUsesPin(uint8 pin) {
      ^~~~~~~~~~~~
In file included from C:\Users\serka\Documents\Arduino\hardware\Arduino_STM32\STM32F1\variants\generic_stm32f103r\wirish\boards.cpp:48:
C:\Users\serka\Documents\Arduino\hardware\Arduino_STM32\STM32F1\cores\maple/boards.h:112:7: note: old declaration 'uint8 boardUsesPin(uint8)'
 uint8 boardUsesPin(uint8 pin);
       ^~~~~~~~~~~~
C:\Users\serka\Documents\Arduino\hardware\Arduino_STM32\STM32F1\variants\generic_stm32f103r\board.cpp:138:1: error: too many initializers for 'const stm32_pin_info'
 };

stevstrong, 2019-09-27:
OK, I see that my repo differs a lot from the original repo, so I suggest you to get the original repo and replace only the SPI lib files from my repo to your local (original) repo.

serkanc, 2019-09-27:
No luck :)
static const spi_pins board_spi_pins[BOARD_NR_SPI] __FLASH__ =

I give up :) Thanks for your kindly reply....

serkanc, 2019-10-03:
I tried fresh from your STM32 Arduino fork, Sdfat from your fork. It compiles for F407 but for F103, BLINK example throw error...
C:\Users\serka\Documents\Arduino\hardware\Arduino_STM32\STM32F1\cores\maple/boards.h:112:7: error: ambiguates old declaration 'uint8 boardUsesPin(uint8)'

 uint8 boardUsesPin(uint8 pin);

       ^

C:\Users\serka\Documents\Arduino\hardware\Arduino_STM32\STM32F1\variants\generic_stm32f103r\board.cpp:138:1: error: too many initializers for 'const stm32_pin_info'

 };

C:\Users\serka\Documents\Arduino\hardware\Arduino_STM32\STM32F1\variants\generic_stm32f103r\board.cpp:138:1: error: too many initializers for 'const stm32_pin_info'

stevstrong, 2019-10-04:
I just made two commits to support SDIO for variant F103R8 in my repo.
I did not test it because I don't have such a board, but it should compile error & warning free.

Please report any further issues relating my repo in my repo: https://github.com/stevstrong/Arduino_STM32/issues

stevstrong, 2019-12-26:
@serkanc ,
have you managed to compile an example error free?

serkanc, 2019-12-26:
Yes, I have compiled and tested. But SdFatEx was faster than Sdio... I have tried with 1 bus line...

stevstrong, 2019-12-26:
@serkanc 
Can you eventually post here the example you compiled?

victorpv, 2019-12-27:
SDIO is slower that SPI with 1 single bit because we can run the SPI port at 72Mhz and the SDIO at 48Mhz. The big advantage comes from using 4 bits.
I also heavily modified the SDIO driver to have all the features of SdFatEx, plus retries in case of errors, which greatly improves reliability when pushing the speed limits for some cards, but I never submitted those because I never cleaned the code.
I did confirm with Bill Greyman that SdFatEx was born out of having to work around hardware issues with the teensy MCU, but when not dealing with those issues he saw no reason to not have the same features in the F1 in the normal SdFat.
If you are happy with 1 data line, then keep using spi, it's indeed faster.

serkanc, 2019-12-27:
// Sdfat config file should be added#if defined(__MK64FX512__) || defined(__MK66FX1M0__) satirina || defined(__STM32F1__) 
// #define INCLUDE_SDIOS 1
// #define ENABLE_EXTENDED_TRANSFER_CLASS 1
// In SdioF1.cpp 
 //if (!cardAcmd(m_rca, ACMD6_XFERTYP, 2))  2 must be 1 for one bit
//  sdio_set_dbus_width(SDIO_CLKCR_WIDBUS_4BIT); dio_set_dbus_width(SDIO_CLKCR_WIDBUS_1BIT); 

#include "SdioF1.h"
#include "SdFat.h"

SdFatSdioEX sdEx;
//SdFatEX sdEx;
File file;
char command[512];
char command1[100][100];

void setup() {
  // put your setup code here, to run once:
  while (Serial.read() != '\n') {
    __asm__("nop\n\t");
  }
  if (!sdEx.begin()) {
    sdEx.initErrorHalt("SdFatSdioEX begin() failed");
  }
  Serial.println("Ok");
  sdEx.ls();
  file.open("PM_xyz-cube.gcode");
  Serial.println(file.fileSize());
  int i;
  unsigned long sizef = 0;
  int sizef1 = 0;
  unsigned long a = micros();
  while (true) {
    /*i = file.fgets(command,100);
      sizef=sizef+i+1;
      if(i == 0) {
      sizef--;
      break;
      }*/
    sizef1 = file.read(command, 512);
    sizef = sizef + sizef1;
    int k = 0;
    int l = 0;
    for (i = 0; i < 512; i++) {
      command1[k][l] = command[i];
      if (command[i] == '\n') {
        Serial.print(command1[k]);
        k++;
        l = 0;
      }
      l++;
    }
    command == "";
    if (sizef1 < 512) {
      break;
    }
  }
  unsigned long b = micros() - a;
  Serial.println(b);
  Serial.println(sizef);
}

void loop() {
  // put your main code here, to run repeatedly:

}

serkanc, 2019-12-27:
SPI was better and stable.

stevstrong, 2019-12-27:
@serkanc ,
thank you for the example.
Maybe we should add it to the repo examples.

serkanc, 2019-12-27:
I will test it 4 bus width, then you can add it.

serkanc, 2019-12-29:
I’m using readline method to read txt file line by line. But it’s too slow against to benchmarks. Can you give me advice for faster line reading of a txt file?