rogerclarkmelbourne/Arduino_STM32 #697: STM32 usb driver not recognised win10

paolovr70, 2019-10-02:
HI thank you for you repository. I used the two batch files for drivers but WIN10 does not recognize the stm32. Message is no associated driver. Any suggestion? Thanks. P.

rtek1000, 2019-10-05:
Windows 10 makes the installer go crazy because of the drivers. Is this the case of unsigned driver? What is the VID / PID that appears in your computer's device manager?

STM32F103C8T6 - Virtual Serial Port:
[image]

paolovr70, 2019-10-08:
> Maple driver

Hi,

sorry but USB nt recognised and driver NOT installed.

Any solution? thanks.

p.

stevstrong, 2019-10-08:
You have to execute [the batch file](https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/master/drivers/win/install_drivers.bat) with admin rights.

paolovr70, 2019-10-15:
Yes. BAT File executed but the problem remains. I tried with 3 different win10 compiters. I am not able to change the driver and use a different one because.the system tell me that it.is already using the best one according to win10. Any suggestion?

rogerclarkmelbourne, 2019-10-15:
I tested this recently on a brand new W10 installation and it worked fine.
Even if the STM32 was already connected.

rtek1000, 2019-10-16:
What board are you using?

Some boards may have a hardware error that may not let your computer recognize USB, a pull-up resistor on the USB data line, see:

Ref: http://amitesh-singh.github.io/stm32/2017/05/27/Overcoming-wrong-pullup-in-blue-pill.html

paolovr70, 2019-10-16:
Thank you, for the reply but in my BOARD R10 ia resistor with the vaue of 152, that is exactly 1.5k.
Any other idea? thanks. p.

adamfxtion, 2019-10-17:
I was have been having problems getting USB to work with recent STM32F103C8T6 system minimum boards, getting 'device not recognised' messages and 'device descriptor request failed' message in device manager (Win 10). I have concluded that while the boards were sold as STM32F103C8T6 system minimum boards and the chips are labelled as STM32F103C8T6, they are in fact something else with STM32 markings. there may of course be versions of the chip with correct markings, not STM32) that that also don't work.
The link below, second page, has more on these boards. While the discussion is about not being able to program STM32's with STlink, it turns out that the reason it doesn't work is because the CPUID was different than that which would be read from a genuine STM32 part.

https://www.eevblog.com/forum/beginners/unexpected-idcode-flashing-bluepill-clone/25/?PHPSESSID=16g9aae4bkem0ei5sru1870i32

I have tested my boards and the configuration bits at 0xE00FFFC0 are a match for Macbeth's when read using using ST-Link Utility, so the manufacturer as read from the chip is not ST if I understand correctly.
I have not found any of the STM32duino code (including bootloaders) to work with these chips, whatever they are. I did successfully get the openCM3 CDC example project to work in VSC/Platformio (after sorting out the STlink programming issue discussed in the link) and Macbeth got the USB to work another way, but as it stands the, it looks as though the current STM32duino code and these chips are incompatible. It also rather looks to me like a large batch of these particular boards with suspect chips is now in circulation, as I have 6 boards obtained in 3 different orders, 2 from UK stock and 4 in 2 lots from China direct (all last couple of months though). I also confirmed that changing the CPU for an STM32 obtained from Farnell on one of these boards fixed the USB issue.

rtek1000, 2019-10-17:
It may also happen that the chip is damaged.

paolovr70, 2019-10-19:
Wow. What a pity! And waste of money.... I BOUGHT 3 ...

rtek1000, 2019-10-19:
Would it be possible to replace only the chip, so as not to lose everything?

adamfxtion, 2019-10-19:
Tricky to know for sure. I did replace the STM32 chip on one board to verify that the STM32 chip was the issue. But it cost me more than 5GBP to buy the part from Farnell, so it doesn't make a lot of economic sense. It will be possible to get them cheaper online of course, but then who knows if you are getting a genuine part?
Then also there is the problem of the actual chip change. Without at the very least a hot air rework station it will probably be virtually  impossible to remove the original STM32 without damage to the board. Resoldering is less of a problem although still not that easy - I usually use solder paste spread with a small blade in situations where I can't use a stencil. I then solder with the rework station on a hot bed and then remove excess with solder wick and a small soldering iron.
But I suspect if you are not sure, the answer is it won't be practical to attempt.

rtek1000, 2019-10-19:
Maybe you can find good chips on generic stlink v2 board, since the use of the USB port is mandatory:

https://github.com/rogerclarkmelbourne/Arduino_STM32/wiki/Programming-an-STM32F103XXX-with-a-generic-%22ST-Link-V2%22-programmer-from-Linux

stevstrong, 2019-10-20:
You may consider to buy in the future "black pills", they cost a bit more than the blue pills but are of higher quality.

paolovr70, 2019-10-20:
Thank you guys! But... Too much complicated for me to unsolder and resolder chip . I will try with the BLACK pill. Otherwise Arduino Nano will remain my favourite small CONTROLLER. ....

paolovr70, 2019-12-23:
FINALKY I tried also with BLACK PILL. USB.connection is not recognised. STM32 is costing too much money for me. (6 non utilised boards...)

rtek1000, 2019-12-23:
😅

stevstrong, 2019-12-23:
You have a driver problem.
Delete the old drivers, and install those of the batch files from the repo with admin rights.

closed 2020-02-06 by rogerclarkmelbourne