rogerclarkmelbourne/Arduino_STM32 #698: usb_midi_is_transmitting() always returns true (usb_midi_device.c)

AurelGTS, 2019-10-02:
Hello,

I m trying to create a midi in and out over usb. I sucessfully received data but i can not transmit... Function usb_midi_is_transmitting() in usb_midi_device.c always return 1 (true). Usb device is well recognised on computer. What am l doing wrong ? I would really, really appreciate yoir help. I spend a lot of time in debug... 😩😩
Thanks,
Best regards, Aurélien

dewhisna, 2019-10-02:
If you want to see a working USB MIDI application using this library that supports receive and transmit, you can checkout my [STM32F103C port of the Altura MIDI Theremin](https://github.com/dewhisna/Altura-MIDI-Theremin/tree/Dewtronics/stm32f103c8t6), specifically the code in [main.cpp](https://github.com/dewhisna/Altura-MIDI-Theremin/blob/Dewtronics/stm32f103c8t6/src/main.cpp).  It uses this Arduino_STM32 repo for its core and library support.  Its project files, though, are setup to use under [PlatformIO](https://platformio.org/) instead of the Arduino IDE, but the code would apply equally for development regardless of the IDE.

That project is mostly the transmit side, as the MIDI Theremin is an "instrument" that uses ultrasonic sensors to detect hand movements/gestures and converts it to MIDI notes to transmit...  maybe it will give you some clues for your project...

AurelGTS, 2019-10-02:
OK, I will see, but I have already used a working project : https://github.com/TheKikGen/USBMidiKliK4x4
I have exactly the same behaviour...

dewhisna, 2019-10-03:
You indicated that it's "recognised on [the] computer", and by that, I assume you mean the computer recognizes the device.  But do you have the endpoint connected on the computer side?  In other words, do you have an application up and running and specifically connected to your USB MIDI device to actually receive the data?

If not, on the STM32 side, it's in-the-queue waiting to transmit (as indicated by the return value you referenced), but nothing on the PC side read the data and so it's "still transmitting".  The Arduino/STM32 side will hang waiting on the connection to be established, if it doesn't have anyone (any application) to send the data to.  Try running an application, like TiMidity++ or KMidimon, etc., on the PC and open the connection to your device.

You will notice in my [MIDI Theremin Code](https://github.com/dewhisna/Altura-MIDI-Theremin/tree/Dewtronics/stm32f103c8t6), I have logic for checking to make sure the USB connection is active and to allow for timeout conditions so that the code on the STM32 doesn't hang if the PC connection isn't actually open or if the user suddenly closes the PC application.

AurelGTS, 2019-10-03:
> You indicated that it's "recognised on [the] computer", and by that, I assume you mean the computer recognizes the device. But do you have the endpoint connected on the computer side? In other [...]
Yes it is, I have a sequencer (Ableton ) connected and it can send message (that I received on STM32). Midi in and midi out devices are selected in Ableton

>
>
>
You will notice in my MIDI Theremin Code, I have logic for checking to make sure the USB connection is active and to allow for timeout conditions so that the code on the STM32 doesn't hang if the PC connection isn't actually open or if the user suddenly closes the PC application.
>
>
I ll check that, thanks.

AurelGTS, 2019-10-03:
OK, got it. I just had a while on !MidiUSB.isConnected() and it works...
Thanks very much dewhisna.
bye

closed 2019-10-03 by AurelGTS