rogerclarkmelbourne/Arduino_STM32 #700: www.stm32duino.com and wiki.stm32duino.com: offline permanently!

rtek1000, 2019-10-04:
I found the explanation as to why the sites are down:

https://www.rogerclark.net/blog-offline-due-to-server-quota-problems/

But unfortunately, many explanations about this core are in these sites, or not?

Would it be possible to back up the content of these sites to make the written answers and tips accessible?

zoomx, 2019-10-04:
The new forum is here
[https://mcu.selfip.com](url)

Serasidis, 2019-10-04:
@zoomx The link must be edited. It does not lead to https://mcu.selfip.com

zoomx, 2019-10-04:
I used the add link button in the editor but it seems that I am not able to use it.

rtek1000, 2019-10-05:
I saw that this site (https://mcu.selfip.com) has only a few posts.

So all the content on the other sites was lost? This is very sad!

rtek1000, 2019-10-05:
https://github.com/stm32duino/Arduino_Core_STM32/issues/577

rtek1000, 2019-10-05:
Due to another issue, I close this, please go to another issue mentioned above, as there is more information.

closed 2019-10-05 by rtek1000

rtek1000, 2019-10-05:
> There is a complete copy of the site on archive.org taken on 15th July:

https://web.archive.org/web/20190715072346/http://www.stm32duino.com/

Source: https://github.com/stm32duino/Arduino_Core_STM32/issues/574