rogerclarkmelbourne/Arduino_STM32 #701: Idle, sleep, deep sleep, shutdown, wakeup?

rtek1000, 2019-10-04:
I found some examples of low power for another core:

> Hardware state
> 
>     Idle mode: low wake-up latency (µs range) (e.g. ARM WFI). Memories and voltage supplies are retained. Minimal power saving mainly on the core itself.
> 
>     sleep mode: low wake-up latency (µs range) (e.g. ARM WFI), Memories and voltage supplies are retained. Minimal power saving mainly on the core itself but higher than idle mode.
> 
>     deep sleep mode: medium latency (ms range), clocks are gated to reduced. Memories and voltage supplies are retained. If supported, Peripherals wake-up is possible (UART, I2C ...).
> 
>     shutdown mode: high wake-up latency (posible hundereds of ms or second timeframe), voltage supplies are cut except always-on domain, memory content are lost and system basically reboots.
> 

Source: https://github.com/stm32duino/STM32LowPower

Is it possible to perform these functions on this core?

rtek1000, 2019-10-04:
After a lot of research, I found how to go into low power mode, but the interruptions and wake up modes still need to be configured:

```
/*
  Blink

  Turns an LED on for one second, then off for one second, repeatedly.

  Most Arduinos have an on-board LED you can control. On the UNO, MEGA and ZERO
  it is attached to digital pin 13, on MKR1000 on pin 6. LED_BUILTIN is set to
  the correct LED pin independent of which board is used.
  If you want to know what pin the on-board LED is connected to on your Arduino
  model, check the Technical Specs of your board at:
  https://www.arduino.cc/en/Main/Products

  modified 8 May 2014
  by Scott Fitzgerald
  modified 2 Sep 2016
  by Arturo Guadalupi
  modified 8 Sep 2016
  by Colby Newman

  This example code is in the public domain.

  http://www.arduino.cc/en/Tutorial/Blink
*/

#include <libmaple/pwr.h> // PWR_BASE ref
#include <libmaple/scb.h> // SCB_BASE ref

#define sleepMode 0
#define stopMode 1
#define standbyMode 2

#define wakeupInterrupt 0
#define wakeupEvent 1

// the setup function runs once when you press reset or power the board
void setup() {
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(LED_BUILTIN, OUTPUT);
}

int counter = 0;

// the loop function runs over and over again forever
void loop() {
  digitalWrite(LED_BUILTIN, LOW);   // turn the LED on (HIGH is the voltage level)
  delay(1000);                       // wait for a second
  digitalWrite(LED_BUILTIN, HIGH);    // turn the LED off by making the voltage LOW
  delay(1000);                       // wait for a second

  if (counter < 10) {
    counter++;
  } else {
    counter = 0;

    pwr_set(standbyMode, wakeupInterrupt);
  }
}

void  pwr_set(int mode, bool wakeUp)
{
  uint32 PWR_CR = PWR_BASE->CR;
  uint32 SCB_SCR = SCB_BASE->SCR;

  if (mode != sleepMode) {
    if (mode == stopMode) {
      PWR_CR |= PWR_CR_LPDS;
    }
    
    PWR_CR |= PWR_CR_PDDS;
    PWR_BASE->CR = PWR_CR;

    SCB_SCR |= SCB_SCR_SLEEPDEEP;
    SCB_BASE->SCR = SCB_SCR;
  }

  if (wakeUp == wakeupInterrupt) {
    asm("WFI"); // Wakeup: Any interrupt
  } else {
    asm("WFE"); // Wakeup: Wakeup event
  }

  // See Table 11. Low-power mode summary
  // RM0008 Reference manual (from STM)
  // PM0056 Programming manual (from STM)
}
```

rtek1000, 2019-10-05:
Something is missing. In the STM32LowPower library it is possible to achieve a few uA consumption.

neryvini, 2020-01-03:
> After a lot of research, I found how to go into low power mode, but the interruptions and wake up modes still need to be configured:
>
> ```
> /*
>   Blink
>
>   Turns an LED on for one second, then off for [...]

That is true, i achieved 5uA using STM32LowPower. Do you achieve any progress with Roger's core?

S4deghN, 2020-03-18:
Hi,
i've had issues for making my stm32f103c go to sleep but i finally figured it out, i post it here so may help others 
`
#include <libmaple/pwr.h> // PWR_BASE ref
#include <libmaple/scb.h> // SCB_BASE ref

  uint32 PWR_CR = PWR_BASE->CR;
  uint32 SCB_SCR = SCB_BASE->SCR;

  RCC_BASE->APB1ENR |= RCC_APB1ENR_PWREN;
    
  PWR_CR |= PWR_CR_PDDS;
  PWR_BASE->CR = PWR_CR;

  SCB_SCR |= SCB_SCR_SLEEPDEEP;
  SCB_BASE->SCR = SCB_SCR;
  
  PWR_BASE->CR |= PWR_CR_CWUF;

  PWR_BASE->CSR |= PWR_CSR_EWUP;

  asm("WFI"); 
`

i found it out by reading the actual core libraries and some help on internet about putting stm32 to sleep 

in this code the device wakes up on external interrupt on pin PA0 (it's set on LOW and wake ups when pulled HIGH, unfortunately it seems like pull up doesn't work when it's put to sleep )

btw it's deep sleep and makes the device reset when woken up by interrupt.

rogerclarkmelbourne, 2020-05-18:
This is not really an issue

Ideally you could submit a PR for a library to support this feature

closed 2020-05-18 by rogerclarkmelbourne