rogerclarkmelbourne/Arduino_STM32 #702: USB not working blue pill boards

adamfxtion, 2019-10-05:
I hope this is not too off topic and doesn't upset anyone, but please forgive if it is. 
I have several 'system minimum STM32F103C8T6' boards bought from ebay and Amazon in 1's and 2s. They are all programmable and seem fine (All PC13 LEDs) excepting that I get no USB on most of them, either using your bootloader or Arduino compiled code. I have one that is a CKS32F103 that works fine. The other 6 have processors all labelled STM32F103, but the quality of the chip labelling and various package variations leads me to believe they may not be genuine STM32s. Anyway, I purchased an STM32 from Farnell in the UK and replaced one of the processors on a non working (USB) board - and the USB started working as expected. I should note that I changed R10 to 1.5K on some of the boards with no apparent difference.
Windows 10 reports no USB descriptor. All the boards that don't work were purchased in the last month and have a 4 pin low profile reset switch rather than the 2 pin rectangular on of the CKS chip board. 
The only possibility that I can see that might be good news is that ST have released an update to the STM32 chip that has broken your code and also the Arduino code. I confess to not trying the STCUbe route as once downloaded it seemed like awfully hard work to get a compiled binary, and as all the chips are labelled STM32's and the code clearly works on a genuine STM32, that I would not go there for the moment. 
Is anyone able to verify if they are having similar problems, if I am just making a silly mistake somewhere, if there might be a batch of broken STM32's in circulation, or if there is a perfectly good reason that new STM32 chips don't work with bootloader and Arduino code and that it will be fixed in the future? Because it will be very difficult to use the boards for any application that needs USB if many have to be discarded as not having working USB. It is not easy to distinguish between boards on Ebay/Amazon. I ordered one with a square reset in the picture, but that isn't what I got and sending £2 boards back is far too much like hard work.
Pictures of 2 working boards (CKS and Resoldered) and 3 non working boards attached.

[image]
[image]
[image]
[image]
[image]

stevstrong, 2019-10-08:
It may happen that some slots from the same seller have the same issue.
In this case the USB connector may be soldered badly. It happened to me, too.
Or you may try simply to apply hot air onto the chip, some weak soldering portion may recover this way.

adamfxtion, 2019-10-08:
The parts were ordered in at least 3 different batches (and there were a couple more I did not bother to photograph and post), so it seems very unlikely to me that it is a single vendor issue. 
Given that I had 6 (have 5) that don't have any working USB, odd dry joints and poor soldering look equally unlikely. In any event, On the board I changed the CPU on, I had checked the USB connections through by toggling the USB IO pins directly from code which worked fine. I also re-soldered the USB connector and checked that it was good. (I have a USB breakout board that allowed me to verify the connectivity).
I am entirely convinced that this is a chip problem. What I am not sure about is if the problem is due to legitimate STM32's having a different spec, either new or old, or more likely given the quality of printing on the package that they are not genuine ST parts. If that is the case, it would be very difficult to buy the boards with any certainty that they might actually function correctly when USB is required.
What I did also fail to mention  but should have is that while programming these with the STlink, I have to press the reset button to get the chip(s) into program mode. Otherwise the STM is not detected. On the 2 boards with working USB, the reset press is not necessary to enter programming mode.

stevstrong, 2019-10-09:
Just because the STLink does not recognise the target, it does not mean that the chip is faulty, it may depend on the debug port setting. If it is disabled, the STLink can only connect to the target shortly after reset, that is "normal".
Is STLink reporting the correct chip?
Have you tried to flash with STLink a simple blinky sketch? Does that work?

rogerclarkmelbourne, 2019-10-09:
Another possibility is one of those fake STM32 chips which may have a different ID

adamfxtion, 2019-10-09:
Thank you very much both for taking the time to reply and your patience. I will do my best to provide accurate information.
First, while I am new to the STM32 chip, I have been developing with many brands of microcontrollers for more than 25 years, both PCB design and software.
I have also been using the Arduino/Platformio-VSC for several years now with Atmel, PIC32, ESP8266 and ESP32 cores. However, I do not have much experience in USB stacks in those microcontrollers (or elsewhere), so am in very new territory with the STM32 generally and the USB peripheral in particular, but should be on firmer footing with everything else.
So Enough of that.

Roger,
If I start with your point, do you mean the 0x410 Device ID shown by the programmers (which all chips report as being), or some other ID to do with the USB port? 

Stev,
It seems on further investigation that the reset was a red herring on my part. All the boards fail to detect the STM32 after a power cycle of the STlink/STM32. Thereafter they all detect fairly reliably. My apologies for that one.
STlink may or may not be reporting the chip correctly as it only states (taken using STM32 Cube programmer output) 

Device             STM32F101/F102/F103 medium-density
Type                MCU
Device ID        0x410
Flash Size      64KB

Log files outputs (on verbose) for both a working and a non working boards (attached below) appear identical.

Unfortunately, I can't test the CKS32 as I sent it to a colleague as part of a prototype PCB we are working on. 

I have no problem programing any of the chips with either Serial (FT232) or STlink programmers. I do not know what the 'debug' port you mention is - if it is the USB Serial Debug port embedded into the Arduino package of the STM32 code that I am uploading, then I have been uploading the same code to all devices. They all claim to be STM32s by markings. The boards are (near) identical and have the LED on PC13, which flashes as expected. I have also bi-passed trying to get the bootloader to work and just compiled and uploaded a simple blinky sketch, both with and without Serial USB port prints. I can program all the boards successfully using either Serial or STlink, but only the board with the Farnell supplied STM32 works with the USB serial port - they all flash the LED as expected. The board with the CKS32 also programs and the USB works with both bootloader and user code as expected.
The STlink programmer Reports STM32 in all cases of STM32 chips. I don't know for the CKS chip.

[STM32 Cube Programmer Log.txt](https://github.com/rogerclarkmelbourne/Arduino_STM32/files/3706763/STM32.Cube.Programmer.Log.txt)

stevstrong, 2019-10-09:
If you are uploading with STM32 Cube, I assume that you are using the official STM core from here: https://github.com/stm32duino/Arduino_Core_STM32.
Is that correct?
If so, then you should post there your problem. This place is reserved for the "Libmaple core", Arduino_STM32.

adamfxtion, 2019-10-09:
I am using the Libmaple core selected in platformio. I used the 'Official' STlink programmer as it seemed like the most reliable source of chip detection/programming, but I have been trying to upload code compiled from Arduino/Platformio/VSC.  I can upload from both these environments successfully as well, but in the interests of ensuring 100% compatibility with the STM32, I used the ST offering to make sure. It also allowed me to suck the working code out of one chip and flash it straight into another (which didn't fix the problem either).

stevstrong, 2019-10-09:
You can try to build and upload from Arduino IDE using the STLink programmer.
Which compiler version do you use?

adamfxtion, 2019-10-09:
Sorry took a while to get all my ducks in a row using the Arduino IDE and STlink rather than Platformio.

Anyway, using a simple blink sketch/Arduino IDE/STlink printing out a count each loop to USB Serial, the board with Farnell sourced STM32 works and prints via USB.
The non working USB board flashes the LED at the correct(ish) rate but I get a USB error:-

A request for the USB device descriptor failed.

This is WIndows 10, Arduino IDE 1.8.8, STM lib 2019.10.9, compiler v4.8.3.

None of these boards worked with USB (All chips marked STM32).
Tried soldering connectors/chips/changed R10 no difference. Always the same error.
Swapped one STM32 for one bought from Farnell and hey presto, it worked fine.

The only possibility I can see is that the STM32's on the boards I bought are not the same as the one bought from Farnell (or obviously the CKS one) as these have working USB with your bootloader code,and with the Arduino IDE and Platformio. With the suspect boards running identical code the USB doesn't work.

SO, in order of likelyhood as I see it, 
Either the chips have broken/missing USB (all of them from different sources on Ebay and Amazon) and will never work (possibly all from the same STM32 batch though)

OR They have potentially working USB but the chip it has differences that breaks the current code base.

OR they are real STM32's but something in the spec of the chip has changed which breaks the existing code.

If it's not the first, then a solution should be possible. What I would ideally like to know is which option is it (and if the first option, then is this just a bad batch in the short term) , although it may just be safer to assume that USB will not work on the boards in all cases, which would be less than ideal but at least something to work with when integrating these boards into a system as I am currently attempting to do.

adamfxtion, 2019-10-16:
I have some further information.
I decided to try using the opencm3 environment (in VSC & PLatformio). When I tried to upload the compiled example, the stlink refused to program the devices, reporting,
hla_swd
Warn : UNEXPECTED idcode: 0x2ba01477
Error: expected 1 of 1: 0x1ba01477
Referencing 
https://www.eevblog.com/forum/beginners/unexpected-idcode-flashing-bluepill-clone/
I changed the stm32f1x.cfg to match the code, the chip programmed successfully and once I copied the 2010 Gareth McMullin <gareth@blacksphere.co.nz> CDC example, I had a USB serial port appear.
Now, just to be clear, these chips are marked as STM32F103C8T6 devices.
Can anyone offer an opinion on what these chips actually are based on the above information (I can of course do some more testing if needed).
And does this mean that if I use the right compiled bootloader.bin (or a custom compile if necessary), and if so which version should it be. I have tried the STM32, Naze and GD versions, but none seem to work. Or is a new version entirely needed for a 'new' chip?

rogerclarkmelbourne, 2019-10-16:
This is off topic.

Please post to STM's own forums, or an OpenCM3 forum etc

closed 2019-10-16 by rogerclarkmelbourne

codewise-nicolas, 2019-12-07:
I just received 2x of the blue pill units similar to your 3rd picture (the one with 2 shallow corner dotcs and one deep dot) and have the same problem with the usb/bootloader.
Tried various bootloaders in the hope it was a clone but no luck.

I hate to re-open closed threads but its been very hard to find anyone else with the same chips I received. Let me know if you have found anything since you opened this issue. Thanks.

rogerclarkmelbourne, 2019-12-07:
This is not a bug in this core. 

Try posting to

https://www.stm32world.com/

codewise-nicolas, 2019-12-07:
You are correct Roger, it is not a bug in the core.
Im hoping adamfxtion will reply and I can then continue this conversation with them in another medium, github removed the means of private messaging quite some time ago.

rogerclarkmelbourne, 2019-12-07:
OK.
Try just flashing the a test sketch that prints to USB serial, using a STLink or via Serial.

I can't be 100% sure, but I think USB Serial is enabled in one of other of those upload modes (perhaps STLink), and see if USB enumerates, when you unplug and plug in again.

If it doesnt, then the USB is probably not connetced

codewise-nicolas, 2019-12-07:
Thanks Roger. After more research and attempts, its clear that the chips I received are re-marked CS32F103's.
In case anyone else stumbles on this issue, you can read more here: https://www.eevblog.com/forum/beginners/unexpected-idcode-flashing-bluepill-clone/25/

The clear tell tale sign is the 3 dots on the chip, and the programmer identifying 128kb of flash when the chip is supposed to only have 64kb.

sovahc, 2023-02-11:
I found this thread while searching "USB not working blue pill" :)

I changed several micro usb cables and one of them worked.
There was some kind of incompatibility in the micro usb connector itself.

With good cable:
$ sudo dmesg -w

/connect blue pill to usb/

[ 2106.854290] usb 1-7.1.1: new full-speed USB device number 17 using xhci_hcd
[ 2106.956272] usb 1-7.1.1: New USB device found, idVendor=0483, idProduct=5740, bcdDevice= 0.00
[ 2106.956285] usb 1-7.1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2106.956290] usb 1-7.1.1: Product: BLUEPILL_F103C8 CDC in FS Mode
[ 2106.956293] usb 1-7.1.1: Manufacturer: STMicroelectronics
[ 2106.956296] usb 1-7.1.1: SerialNumber: XXXXXXXXXXX
[ 2106.964490] cdc_acm 1-7.1.1:1.0: ttyACM0: USB ACM device

With wrong cable:

silence