rogerclarkmelbourne/Arduino_STM32 #704: Wrong values from I2C sensor

bergos, 2019-10-06:
I'm getting wrong values from my BME280 sensor using the latest version of Adafruit library with the master branch of Arduino_STM32. The initialization of the sensor works, but the values form the `read*` methods don't make any sense. I saw a lot of I2C related changes in the last time by @dewhisna and @stevstrong, so I tried with a version from July, before all the changes were added and it worked again (a3a56866505eeca4b49fc435700dcd4bd5540c8c). I'm happy to help to narrow down the changes which cause the problem, but there are a lot of commits in the meantime. If you could give me a list of commits which are worth testing, I can test them and report where the problems starts to show up.

Here the sketch I'm using:
```C
#include <Adafruit_Sensor.h>
#include <Adafruit_BME280.h>
#include <Arduino.h>
#include <Wire.h>

Adafruit_BME280 bmeSensor;

void setup() {
  Serial.begin(115200);

  delay(5000);

  Serial.print("init BME280...");
  if (!bmeSensor.begin(0x76)) {
    Serial.println("failed");
  } else {
    Serial.println("done");
  }
}

void loop() {
  Serial.print("temperature: ");
  Serial.println(bmeSensor.readTemperature());

  Serial.print("humidity: ");
  Serial.println(bmeSensor.readHumidity());

  delay(1000);
}
```

ssieb, 2019-10-06:
An example of data values that you are getting and what you expect might be helpful.

bergos, 2019-10-06:
@ssieb thanks for the hint. I added example output for both version to the first post.

dewhisna, 2019-10-06:
@bergos -- make sure your code for the BME280 is correct.  There's a lot of broken libraries out there that don't conform to the Bosch specs -- even official Adafruit libraries.  In fact, 2-1/2 years ago, I submitted a [PR to Adafruit to fix the Python version of the Adafruit BME280 library](https://github.com/adafruit/Adafruit_Python_BME280/pull/6). 

I haven't compared their Arduino (or other non-Python) variants, but it's very very likely they have similar problems that have never been fixed.  Before diving into the I2C code here, I first suggest you compare the BME280 library you are using very carefully against the [Bosch BME280 datasheet](https://www.bosch-sensortec.com/bst/products/all_products/bme280) at [https://ae-bst.resource.bosch.com/media/_tech/media/datasheets/BST-BME280-DS002.pdf](https://ae-bst.resource.bosch.com/media/_tech/media/datasheets/BST-BME280-DS002.pdf).

The specific problem I think you are seeing is the first one on the list for that PR on the Python version: 
> Fixed the inconsistent data read issue when reading values from the BME280 by doing the read as a single burst transfer as detailed in the Bosch Datasheet instead of register-by-register reads. This mainly affected the humidity sensor reading, which previously caused it to return erroneous values.

In other words, you have to do the data read as a single burst transfer as detailed in the Bosch spec rather than register-by-register.  My bet is that it wasn't actually working completely correctly for you with the previous I2C code either and would occasionally yield wrong readings there too -- perhaps not as often, but probably was a problem you didn't realize you had.

bergos, 2019-10-06:
@dewhisna I can compare the spec to the library in the next days, but I can tell you already that I can reproduce the error case and the good case all the time, tested with different bluepill boards and different BME280 sensors. Also I'm running multiple BME280 + bluepill boards for weeks now, writing values every 1min to a database and generating charts out of it. As you can see in the example data in the updated post, the values are very different, so I would immediately notice such an error in my charts. That's why I would rule out that this error occurred in the past.

dewhisna, 2019-10-06:
Is this [Adafruit_BME280 Library](https://github.com/adafruit/Adafruit_BME280_Library) the one that you are using??  If so, at first-glance (without too deep of a dive into the code), it appears that it too is broken like the Adafruit Python version was.  Specifically, the three values (pressure, temperature, and humidity) should be read via a single burst read.  See this commit: [https://github.com/adafruit/Adafruit_Python_BME280/commit/a0586e4e4f6df701b6018a4f8128813b36d6d456](https://github.com/adafruit/Adafruit_Python_BME280/commit/a0586e4e4f6df701b6018a4f8128813b36d6d456).  And compare that to the read code in the C++ version of [Adafruit_BME280 Library](https://github.com/adafruit/Adafruit_BME280_Library).

It's likely that the old I2C here behaved slightly different with regard to the timing of the start/stop signals on the bus and in some cases caused the BME280 to treat it more like a burst read.  Other than that the Bosch spec says that they must be read in a single burst, I'm not totally sure what timing change causes the data to get skewed.  Even their broken Python code would sometimes return the correct results.  And just because it "happened to work" before doesn't mean it was actually working correctly.

If you change the code in the library you are now using to do the read of all 3 values as a single burst like the Bosch spec says to do and you still see issues, then perhaps one should look closer at the I2C code.

dewhisna, 2019-10-06:
The part I'm referring to in the Bosch spec is on page 23:
>To read out data after a conversion, it is strongly recommended to use a burst read and not address every register individually. This will prevent a possible mix-up of bytes belonging to different measurements and reduce interface traffic. Note that in I²C mode, even when pressure was not measured, reading the unused registers is faster than reading temperature and humidity data separately.

...

>4.1 **Data register shadowing** : In normal mode, the timing of measurements is not necessarily synchronized to the readout by the user. This means that new measurement results may become available while the user is reading the results from the previous measurement. In this case, shadowing is performed in order to guarantee data consistency. Shadowing will only work if all data registers are read in a single burst read. Therefore, the user must use burst reads if he does not synchronize data readout with the measurement cycle. Using several independent read commands may result in inconsistent data. If a new measurement is finished and the data registers are still being read, the new measurement results are transferred into shadow data registers. The content of shadow registers is transferred into data registers as soon as the user ends the burst read, even if not all data registers were read. The end of the burst read is marked by the rising edge of CSB pin in SPI case or by the recognition of a stop condition in I2C case. After the end of the burst read, all user data registers are updated at once.

dewhisna, 2019-10-07:
And another thing you can try somewhat easily is to switch from the `Wire` library to the new `WireSlave` library, since it was recently reworked to support both master and slave operation correctly and fix some other issues.  Just change the `#include <Wire.h>` to `#include <Wire_slave.h>` at the top of your sketch and swap the order of the includes to make sure the Arduino IDE finds the `WireSlave` library ahead of the Adafruit library so that the Adafruit library will see the correct `Wire` objects:

```C++
#include <Arduino.h>
#include <Wire_slave.h>
#include <Adafruit_Sensor.h>
#include <Adafruit_BME280.h>
```

I won't guarantee that will fix it, though, since the Adafruit library doesn't conform to Bosch spec and returning bizarre readings is the same issue I originally had when I ended up reworking their Python version of that library and that was on a Raspberry Pi even (totally unrelated to this Arduino_STM32 core).

If you are unable to get the Adafruit library straightened out, I do have working C/C++ code for the BME280 that I could share.  But it's written for Linux with `i2c-dev`.  Though it should be somewhat straightforward to swap the `i2c-dev` calls for `WireSlave` calls and run on the STM32F1.

bergos, 2019-10-07:
@dewhisna 

I tried `WireSlave`, but already the `.begin()` method of the library never returns when the alternative Wire library is used. I changed the includes like in your comment, but didn't spend much time investigating the problem as I wanted to rule out it's the missing burst transfer.

So I tried the burst transfer. I hope I got it right. [Here](https://github.com/bergos/Adafruit_BME280_Library/commit/45b90c10a68753f1e5d18bcbf123163b5d9016c8) the commit with the changes I made to the fork of the Adafruit library. All 8 bytes are read at once in the `.readAll` method. The rest of the logic is copied from the other methods, besides reading the values from the 8 byte data block. If that's what you meant, then the burst transfer didn't fix it. I still get wrong values in the master and correct values in the older version. Also, as far as I understand the spec from Bosch, mixing bytes from two measures, but still the same offset, is the worst thing that could happen. I didn't check the complex calculation to the final value, but I expect the values wouldn't be so consistent wrong if the temperature, pressure and humidity didn't change much.

dewhisna, 2019-10-08:
If it doesn't even return from `begin()` with the `WireSlave` library (since it's currently the better of the two libraries), I'm suspecting you have a wiring issue on either the SDA or SCL line, as there is reset logic that's part of `begin()` (based on the ST errata datasheet regarding bus reset) in that library where it releases those lines back to GPIO mode and changes their state and waits for it to fully drive the outputs low and then back high to clear the I/O filters on those pins that can get stuck.

If the CPU can't drive both lines fully low and then fully back high, it can get stuck waiting for their state to change.  But that would only happen if you had a short to either V+ or GND or something on those lines (or the lines shorted together).  So I'm very suspicious you have some sort of hardware issue.  What value pull-up resistors do you have installed on them??

I have a bluepill on a special breakout board here and have an I2C BME280 module.  I'm in the process of trying to replicate your issue.  If it fails here, I'll see if I can figure out what's up with the I2C code.  But if it works, it will point back to your hardware setup.  I should know either later tonight or perhaps tomorrow evening.

dewhisna, 2019-10-08:
I think I've recreated the behavior you are seeing, but looking at the raw data it's reading over I2C, they aren't fundamentally different from each other -- meaning it's not in the basic I2C code itself.

In your loop where you are reading the data in `readAll`, I changed this:
https://github.com/bergos/Adafruit_BME280_Library/blob/45b90c10a68753f1e5d18bcbf123163b5d9016c8/Adafruit_BME280.cpp#L542-L544
to this:
```C++
    Serial.print("Bytes: ");
    for (int i=0; i<8; i++) {
      data[i] = _wire->read();
      Serial.print(' ');
      Serial.print(data[i], 16);
    }
    Serial.println("");
```

So I could see the raw bytes.  I was expecting to see something fundamentally different in the data between the old/new I2C code or even `Wire` vs `WireSlave`.  But the raw data looks basically the same.

With the new I2C code, I get the following with the messed up results like you are seeing.:

```
Bytes:  53 BA E0 7B 6A 50 72 64
temperature: -0.31
pressure: 67454.16
humidity: 0.00
Bytes:  53 B9 E0 7B 68 70 72 70
temperature: -0.31
pressure: 67455.69
humidity: 0.00
Bytes:  53 BA 70 7B 68 0 72 72
temperature: -0.31
pressure: 67454.83
humidity: 0.00
Bytes:  53 B9 C0 7B 69 60 72 6D
temperature: -0.31
pressure: 67455.88
humidity: 0.00
Bytes:  53 BA B0 7B 69 E0 72 70
temperature: -0.31
pressure: 67454.45
humidity: 0.00
```

With the old I2C code, I get:
```
Bytes:  53 BB A0 7B 6B 60 72 7A
temperature: 17.80
pressure: 99068.56
humidity: 49.07
Bytes:  53 BB 50 7B 6C 70 72 78                                                 
temperature: 17.80                                                             
pressure: 99070.19                                                             
humidity: 49.06                                                                
Bytes:  53 BB 40 7B 6C 50 72 73
temperature: 17.80
pressure: 99070.35
humidity: 49.04
Bytes:  53 BC 50 7B 6D 40 72 82
temperature: 17.81
pressure: 99068.34
humidity: 49.11
Bytes:  53 BB 80 7B 6A C0 72 5B                                                 
temperature: 17.79                                                             
pressure: 99068.48
humidity: 48.91
Bytes:  53 BA B0 7B 6B 30 72 67
temperature: 17.80
pressure: 99071.06
humidity: 48.97
Bytes:  53 BA D0 7B 6B A0 72 58
temperature: 17.80
pressure: 99071.14
humidity: 48.88
Bytes:  53 BA E0 7B 6C 60 72 68
temperature: 17.80
pressure: 99071.35
humidity: 48.97
```

I was expecting to see a shifted byte or dropped byte or something different, but the output from both sets of I2C code is basically the same.  I do see some random `Serial.print` goofiness of it dropping some of the serial print data.  Makes me wonder if there's an issue with printing the floats or something (but probably not related to the issue at hand).

The new I2C code appears to be functioning OK...  With raw data that similar, there's only one way the values can be that different and that's if the Adafruit library misread the scaling values from the BME280 during its init logic.  That's the only place it can be.

If the scaling values it reads are skewed, all of the readings thereafter will be wrong.  So at the moment, it still looks like it's an Adafruit library bug instead of I2C -- as the bytes it's reading from I2C with the new I2C code look the same as the old I2C code...  So try looking in the init logic ... I'll do some more digging here as well...

I'm betting, and this is just a guess at this point, that there needs to be a delay added or perhaps an extra read cycle with the BME280 from the point of first initialization until the first set of scaling values are read.  The new I2C code has logic to reset the I2C peripheral, including slave devices on the bus, to fix some ST errata conditions with I/O filters, and it's possible it's prolonging the power-up logic of the BME280 until it's ready to provide the calibration values.  So that's where we need to look.

dewhisna, 2019-10-08:
OK...  looks like the second byte in all of the 2-byte reads (`read16_LE` and `readS16_LE`) in the `readCoefficients` function (that is, the upper or most-significant byte) is coming back as `FF`.

So now begs the question -- how can `requestFrom` with a size of 8 work, but a `requestFrom` with a size of 2 not?  It would seem that the BME280 might have NAK'd it or somehow went bus-idle for that second byte, which would result in an `FF` value, but why?

I will have to do some digging on this to figure out where the problem is at.  At "2 bytes to go" on the read, the I2C IRQ does transition from RXNE mode to BTF mode.  And this one is the special `POS` bit case for the `CR1` register, as stated in the ST reference manual:
>Note: The POS bit is used when the procedure for reception of 2 bytes (see Method 2:
transfer sequence diagram for master receiver when N=2) is followed. It must be
configured before data reception starts. In this case, to NACK the 2nd byte, the ACK bit
must be cleared just after ADDR is cleared. To check the 2nd byte as PEC, the PEC bit
must be set during the ADDR stretch event after configuring the POS bit.

That would be the code path it's on.  But I don't know what is different about the BME280 compared to other devices I've tested, as I'm not seeing issues with the other devices I've tried...  Hmmm...  Give me some time to investigate...

dewhisna, 2019-10-08:
@bergos when you get a chance, please pull my PR #706 locally and give it try.  It fixes it here on my BME280 setup.  I had missed the operative keyword "after" in the ST spec where it says in the footnotes: "the ACK bit must be cleared just after ADDR is cleared".  The order of clearing the ACK bit relative to reading the SR2 register (which clears the ADDR) was wrong in the code.

This only affected the 2-byte read path.  Apparently when reworking the I2C code, all of my test cases here for master-reading involved lengths other than 2-bytes.  I thought for sure that I had 2-byte reads, but they were mainly 1-byte, 3-bytes, and 5-bytes.  So thanks for catching this one.

closed 2019-10-08 by stevstrong

dewhisna, 2019-10-09:
@bergos  Glad everything's working again for you.  Thanks for finding the bug.  Sorry I didn't believe it to be an I2C issue initially.  I have spent a lot of time these last few weeks testing the I2C code and thought all such cases had been tested.  That and I remember the issues I had 2-1/2 years ago with the BME280 and weird readings from it.

This is just yet another example of how insane this I2C peripheral is.  I'm not sure what ST was thinking when they designed it.  You have to read every little footnote and side-comment in the datasheet very carefully.  When it says that you have to set this bit after reading such-and-such register, it really means it.  If you set it before reading the register, it doesn't work.

PR #706 (2019-10-08): Fix I2C master 2-byte read case
This fixes #704 

The I2C master mode 2-byte read was incorrectly NAKing prematurely causing the second byte of all 2-byte reads to be returned as `FF`.

As per the ST datasheet for the CR1 register when using read Method 2, it's very important to clear the ACK bit AFTER clearing the ADDR (not before the ADDR is cleared!):

>Note: The POS bit is used when the procedure for reception of 2 bytes (see Method 2:
transfer sequence diagram for master receiver when N=2) is followed. It must be
configured before data reception starts. In this case, to NACK the 2nd byte, [... cut]