rogerclarkmelbourne/Arduino_STM32 #707: HardwareTimer wrong comput

bonarm, 2019-10-08:
I found the HardwareTimer's setPrescaleFactor and/or setCompare and setOverflow maybe wrong. In the sample "HardwareTimerPWMInput", there is:
*********************************************************
const uint8_t pwmPulse = 20; // ms
const uint8_t pwmPeriod = 50; // ms
.......
Timer3.setCompare(TIMER_CH1, pwmPulse*1000);
Timer3.setOverflow(pwmPeriod*1000);
.....
*********************************************************

Looks no problem, PWM output should be 1/50ms=20Hz. But I messured the output freq, it is 2Hz.

The reason I tested this sample is I am working on continuous ADC with timer(3) trigged. For testing, I used another timer(2) work in PWM. I found the ADC times could not match the PWM. So one of them must be wrong. and checking PWM is easy.

stevstrong, 2019-10-09:
And to which value do you set the prescaler in your example?

bonarm, 2019-10-10:
I've found the reason. Add:
Timer3.resume();

Then it worked properly.

Please close the issue.  Thanks.

closed 2019-10-10 by bonarm