rogerclarkmelbourne/Arduino_STM32 #710: Change I2C pin in stm32f103c series

rifqimuzhafar1111, 2019-10-27:
hello, i have a problem in using PB8 and PB9 as i2c pin (scl sda). as i know that the default i2c pin is in PB6 and PB7, but i have used that pin to read encoder, and then i want to change the i2c pin to PB8 and PB9.
thank you, i hope anyone can help me.

dewhisna, 2019-10-27:
The "short" answer is to just set `I2C_REMAP` on the port's flags.

It's technically "deprecated" in the core code, since it's intended to be part of the board variant setup.  That is, it's intended that you'd create a complete set of files for your board that define all the I/O pins and what maps to what, and use that board variant instead of the "generic" one.  But the functionality is still in place for it and you can make it work without doing all of that.

The first question is which Wire library are you using?  Are you using `Wire` or `WireSlave`?  Neither library has an exposed function for it.  It's part of the `FLAGS` for the port that can be set with the call to the `TwoWire` constructor.  Just "OR" the flags with `I2C_REMAP`.

The difference in the two libraries (and this is part of the private implementation difference between them) is that `Wire` has its constructor as `public`.  And `WireSlave` is protected.  And their signatures are slightly different.  `Wire` takes the I2C port index and `WireSlave` takes the pointer to the port device structure.

This means for the `Wire` library, you can simply create your "Wire" object directly and pass the flag to the constructor.  And if you are using `WireSlave`, you'll have to override the `TwoWire` class so that you can call its constructor (i.e. a "sneaky" way to get public access to its protected constructor) and set the flag either in the override's constructor or the creation of your object.

And either way, you'll have to use a different object name other than "Wire" for the I2C, as you'll need an object created with different options than the default.  The default "Wire" object that's constructed automatically for you maps to the default location and settings.  Here's how to do it with both libraries:

**Wire Library:**
```C++
#include <Wire.h>

TwoWire myWire(1, I2C_REMAP);

void setup()
{
    ...

    myWire.begin();

    ...
}

void loop()
{
    ...

    // Replace this example with your actual code, but use 'myWire' instead of 'Wire':
    myWire.beginTransmission(0x3C);    // begin transmission to device 0x3C
    myWire.write("test");
    myWire.endTransmisssion();

    ...
}
```

**WireSlave Library:**
```C++
#include <Wire_slave.h>

class MyTwoWire : public TwoWire
{
public:
    MyTwoWire(i2c_dev* i2cDevice, uint32_t flags = I2C_SLAVE_GENERAL_CALL, uint32_t frequencyHz = 100000)
        : TwoWire(dev, flags | I2C_REMAP, frequencyHz)
    { }
};
MyTwoWire myWire(I2C1);

void setup()
{
    ...

    myWire.begin();

    ...
}

void loop()
{
    ....

    // Replace this example with your actual code, but use 'myWire' instead of 'Wire':
    myWire.beginTransmission(0x3C);    // begin transmission to device 0x3C
    myWire.write("test");
    myWire.endTransmisssion();

    ...
}
```

You can also do it without using either "Wire" library, if you wanted your own low-level implementation of the `TwoWire` functionality.  For that method, simply set `I2C_REMAP` in the flags for your call to either `i2c_master_enable` or `i2c_slave_enable`.

dewhisna, 2019-10-27:
Actually, I just realized that what I wrote won't work for the `WireSlave` library as it currently stands, since that library's `begin` function only lets `I2C_FAST_MODE` and `I2C_DUTY_16_9` flags through, due to how it's using the special flags for slave-mode.  It's a part of the `WireSlave` library rewrite that I haven't completed yet.

So for now, use either the `Wire` library or a lower-level implementation for port remapping.  I will see if I can add that to the code and get a pull-request to update `WireSlave` to make it a little more complete.

dewhisna, 2019-10-27:
And there's yet another way you could do it that would work with either library and should work with the existing `Wire` object without having to construct your own.  And that is, you can manually set the GPIO pins in the `I2C1` structure prior to calling `begin`.  For example:

```C++
void setup()
{
    ...

    I2C1->sda_pin = 9;
    I2C1->scl_pin = 8;
    Wire.begin();

    ....
}
```

This is all the deprecated `I2C_REMAP` flag is doing anyway.  The I2C GPIO init functions will handle the alternate function remap setting correctly based off of the pin numbers and is technically "less deprecated" in terms of the I2C libraries.

Normally this would be done in your board variant code (when creating a custom board definition), as the above code makes it look like you could set the pin numbers to anything you want, which isn't true.  If you set them to anything other than pins PB8 and PB9, then it will default back to PB6 and PB7

stevstrong, 2019-10-27:
Donna, did you mean
```
    I2C1->sda_pin = PB9;
    I2C1->scl_pin = PB8;
```
Otherwise the pins 8 and 9 are equivalent to PA8 and PA9.

dewhisna, 2019-10-27:
No, I meant 8 and 9.  The "pin numbers" in that structure are relative to the `gpio_dev` entry that it's on instead of overall device pin numbers.  In other words, it's the "bit" on the port (not the actual pin number).  From the `i2c_dev` struct definition in `i2c_common.h`:

```C++
    /**
     * @brief SDA GPIO device (but see .gpio_port).
     */
    struct gpio_dev *sda_port;

    /**
     * @brief SCL GPIO device (but see .gpio_port).
     */
    struct gpio_dev *scl_port;

    uint16 msgs_left;           /**< Messages left */
    uint8 sda_pin;              /**< SDA bit on gpio_port */
    uint8 scl_pin;              /**< SCL bit on gpio_port */
```

The `sda_port` and `scl_port` members [and/or the deprecated `gpio_port` member] will be pointing to `gpiob` (which are set at the top of `i2c_f1.c`) and the `sda_pin` and `scl_pin` are the bits on that port.

In fact, the code in `series/i2c.h` that handles the deprecated `I2C_REMAP` flag is where I cut-and-pasted that 8 and 9 from, which reads:
```C++
static inline void _i2c_handle_remap(i2c_dev *dev, uint32 flags) {
    if ((dev == I2C1) && (flags & I2C_REMAP)) {
        afio_remap(AFIO_REMAP_I2C1);
        I2C1->sda_pin = 9;
        I2C1->scl_pin = 8;
    }
}
```

So it should be 8 and 9 and not PB8 and PB9.

Now, if you wanted to keep it as PB8 and PB9 to be less confusing to read, you could write it as:

```C++
        I2C1->sda_pin = PIN_MAP[PB9].gpio_bit;
        I2C1->scl_pin = PIN_MAP[PB8].gpio_bit;
```

The `sda_pin` and `scl_pin` members of `i2c_dev` should have probably been called `sda_pinbit` and `scl_pinbit` or something.

rifqimuzhafar1111, 2019-10-28:
hi dewhisna,
previously I used the Wire_slave.h library for my program, then I also tried your advice and this really works. I added this part in the void setup. 
https://github.com/rogerclarkmelbourne/Arduino_STM32/issues/710#issuecomment-546712010
thank you for helping.

rifqimuzhafar1111, 2019-10-28:
> And there's yet another way you could do it that would work with either library and should work with the existing `Wire` object without having to construct your own. And that is, you can manually set the GPIO pins in the `I2C1` structure prior to calling `begin`. For example:
> 
> ```c++
> void setup()
> {
>     ...
> 
>     I2C1->sda_pin = 9;
>     I2C1->scl_pin = 8;
>     Wire.begin();
> 
>     ....
> }
> ```
> 
> This is all the deprecated `I2C_REMAP` flag is doing anyway. The I2C GPIO init functions will handle the alternate function remap setting correctly based off of the pin numbers and is technically "less deprecated" in terms of the I2C libraries.
> 
> Normally this would be done in your board variant code (when creating a custom board definition), as the above code makes it look like you could set the pin numbers to anything you want, which isn't true. If you set them to anything other than pins PB8 and PB9, then it will default back to the I2C on PB6 and PB7 (but without configuring the GPIOs correctly).

hi dewhisna,
previously i used the Wire_slave.h library for my program, and then i also tried your advice and this really works. i added this part in the void setup.
thankyou for helping.

closed 2019-10-28 by rifqimuzhafar1111