rogerclarkmelbourne/Arduino_STM32 #711: PA12 pin not toggling on stm32f103c series

Broman3100, 2019-10-28:
The digitalWrite(PA12, LOW/0) function for PA12 pin doesn't work on generic stm32f103 board, (f103c8), I know it's not just my board problem since I used keil and it worked just fine so I figured the problem is in stmduino.

stevstrong, 2019-10-28:
Please provide an example sketch for Arduino which you have tested.

Broman3100, 2019-10-28:
after noticing the problem I tested it on another simple sketch:
[image]
The PC13 led blinked without problem where PA12 stayed high all the time

fpistm, 2019-10-28:
Probably USB is enabled...

Broman3100, 2019-10-28:
I'd assume so if it's default, how do I disable it?

stevstrong, 2019-10-28:
I think with Serial.end();

stevstrong, 2019-12-19:
I will close this as there was no update long time.

closed 2019-12-19 by stevstrong

Broman3100, 2019-12-20:
I forgot to say it here, it was usb causing the problem and Serial.end() solved it.