rogerclarkmelbourne/Arduino_STM32 #713: Input Capture Mode using stm32f407 DIY_F407VG

albaradies, 2019-11-09:
I'm trying to run code that works fine on bluebill stm32f103 on stm32f407 DIY_F407VG but it not working the compiler is fine and the upload but there is no capture from the RC the code is attached below ? can someone help me plz 
I try to keep pin PA0 on timer5 but it still the same problem how to fix it plz 

`int32_t channel_1_start, channel_1_stop, channel_1;

void setup() {
  Serial.begin(1000000);
  pinMode(PA0, INPUT);
  delay(250);
  gpio_set_af_mode(PA0, 1);
  Timer2.attachCompare1Interrupt(handler_channel_1);
  TIMER2_BASE->CR1 = TIMER_CR1_CEN;
  TIMER2_BASE->CR2 = 0;
  TIMER2_BASE->SMCR = 0;
  TIMER2_BASE->DIER = TIMER_DIER_CC1IE;
  TIMER2_BASE->EGR = 0;
  TIMER2_BASE->CCMR1 = TIMER_CCMR1_CC1S_INPUT_TI1;
  TIMER2_BASE->CCMR2 = 0;
  TIMER2_BASE->CCER = TIMER_CCER_CC1E;
  TIMER2_BASE->PSC = 71;
  TIMER2_BASE->ARR = 0xFFFF;
  TIMER2_BASE->DCR = 0;
}

void loop() {
  delay(1000);
  Serial.println(channel_1);
}

void handler_channel_1(void) {
  if (0b1 & GPIOA_BASE->IDR) {
    channel_1_start = TIMER2_BASE->CCR1;
    TIMER2_BASE->CCER |= TIMER_CCER_CC1P;
  }
  else {
    channel_1 = TIMER2_BASE->CCR1 - channel_1_start;
    if (channel_1 < 0)channel_1 += 0xFFFF;
    TIMER2_BASE->CCER &= ~TIMER_CCER_CC1P;
  }
}

`

albaradies, 2019-11-09:
I Solve it using this adding code
`gpio_set_mode(PA0,GPIO_AF_INPUT);       
  gpio_set_mode(PA1,GPIO_AF_INPUT);       
  gpio_set_mode(PA6,GPIO_AF_INPUT);       
  gpio_set_mode(PA7,GPIO_AF_INPUT);       
  gpio_set_mode(PB0,GPIO_AF_INPUT);       
  gpio_set_mode(PB1,GPIO_AF_INPUT);       
  gpio_set_mode(PB6,GPIO_AF_OUTPUT_PP);       
  gpio_set_mode(PB7,GPIO_AF_OUTPUT_PP);       
  gpio_set_mode(PB8,GPIO_AF_OUTPUT_PP);       
  gpio_set_mode(PB9,GPIO_AF_OUTPUT_PP);       

  gpio_set_af_mode(PA0, GPIO_AFMODE_TIM1_2);
  gpio_set_af_mode(PA1, GPIO_AFMODE_TIM1_2);
  gpio_set_af_mode(PA6, GPIO_AFMODE_TIM3_5);
  gpio_set_af_mode(PA7, GPIO_AFMODE_TIM3_5);
  gpio_set_af_mode(PB0, GPIO_AFMODE_TIM3_5);
  gpio_set_af_mode(PB1, GPIO_AFMODE_TIM3_5);
  gpio_set_af_mode(PB6, GPIO_AFMODE_TIM3_5);
  gpio_set_af_mode(PB7, GPIO_AFMODE_TIM3_5);
  gpio_set_af_mode(PB8, GPIO_AFMODE_TIM3_5);
  gpio_set_af_mode(PB9, GPIO_AFMODE_TIM3_5);`

stevstrong, 2019-12-19:
It is known that F4 requires the AF_MODE setting for pins.

closed 2019-12-19 by stevstrong