rogerclarkmelbourne/Arduino_STM32 #715: Is there any way to change the statck size?

Liuyiming11, 2019-11-15:
I want to change the chip's stack size ,is there any way to do this?

dewhisna, 2019-11-15:
Not sure what you are trying to do, but there is no "stack size".  The stack starts at the top of RAM and grows downward.  The heap starts at the end of global variables and grows upward.  The two grow until they collide and you run out of RAM and they start corrupting each other and your code begins crashing or behaving erratically.  If you need more stack space, you'll need to use less heap and/or less global variables and/or buy a larger scale STM32 chip that has more RAM.  Or use less nested function calls and less local variables in functions so you don't need as much stack to begin with.

If you want to experiment around with how memory is configured and code is linked, locate the .ld linker script files (and their .inc include files) for your board/variant and edit them.  Or better yet, copy them to your project, and use a local board override and/or linker compile flag to load your linker script instead of the framework's.  Just be careful not to break init/fini layout or the interrupt vector table or you'll end up spending many hours debugging things for naught.  In other words, there isn't much to edit/tweak there that's likely to help you.

The only "quirk" I've found so far in the linker scripts is that for some reason `_lm_heap_end` doesn't get set to `__msp_init` like I'd expect, but rather to the top of address space, meaning it doesn't technically know when you are out of RAM when doing a `malloc`.  But that doesn't really matter much, because if the heap did ever grow even to `__msp_init`, you'd already crash due to not having any stack left anyway.

There's a "sneaky" workaround that I use on my projects that fixes that one by setting the `CONFIG_HEAP_END` define.  For example, on the STM32F103CB micro, adding `-DCONFIG_HEAP_END='((void *)0x20004800)'` to the build flags will set the `_lm_heap_end` extern (used by `syscalls.c` and specifically the `_sbrk` function) to the correct `__msp_init` address for that micro.  If you wanted your `malloc` to fail and return NULL pointers before running totally and completely out of RAM, you could just set that define to a different (lower) address than `__msp_init`.

rogerclarkmelbourne, 2019-11-15:
This also begs the question, what is the OP doing thats consuming so much stack, or is their application really running out of stack

The OP needs to provide a lot more information ;-)

dewhisna, 2019-11-15:
@rogerclarkmelbourne  yeah, I mostly provided a complete answer to benefit the wider audience -- particularly the trick for setting the upper heap address.  Hopefully Google will index it and make it easy for others to find. :-)

stevstrong, 2019-12-26:
I think Donna provided a complete answer to this issue, so I will close this.
Reducing stack usage is possible by reducing dynamically allocated memory in favor of static allocation.

closed 2019-12-26 by stevstrong