rogerclarkmelbourne/Arduino_STM32 #716: Virus detect by Kaspersky

Boia11, 2019-11-18:
Hi,
the file `drivers/win/wdi-simple.exe` detected like virus from Kasperky: 

VirusTotal: [https://www.virustotal.com/gui/file/2baf73145c8800a3a60a3dd50ac589337f18ac7c31fdc8c70d525ca0ff41ce81/detection](https://www.virustotal.com/gui/file/2baf73145c8800a3a60a3dd50ac589337f18ac7c31fdc8c70d525ca0ff41ce81/detection)

rogerclarkmelbourne, 2019-11-18:
The file has not been updated for 3 years.

If Kasperky now decides its a virus, you'd have to question why its not been detected by the thousands of people that have downloaded the file in the last 3 years

Basically, the world of virus scanning is now broken.

Virus companies regularly tag files as viruses which are not.    AKA False Positvies

Boia11, 2019-11-18:
File submitted for analysis on Kaspersky VirusDesk. I'm waiting the mail report by Kaspersky Lab HQ

rogerclarkmelbourne, 2019-11-19:
OK

stevstrong, 2019-12-19:
@Boia11 , any update on this?

Boia11, 2019-12-20:
> 
> 
> @Boia11 , any update on this?

Nope, nothing

stevstrong, 2019-12-21:
Then I will close this, as not relevant for the functionality of this core.

closed 2019-12-21 by stevstrong

bogtogus, 2024-09-14:
This file is detected as virus using [Virustotal](https://www.virustotal.com/gui/file/2baf73145c8800a3a60a3dd50ac589337f18ac7c31fdc8c70d525ca0ff41ce81/detection) site. "Thousands of people" does not seem like good argument as source files.

bogtogus, 2024-09-14:
This file is detected as virus using [Virustotal](https://www.virustotal.com/gui/file/2baf73145c8800a3a60a3dd50ac589337f18ac7c31fdc8c70d525ca0ff41ce81/detection) site. "Thousands of people" does not seem like as good argument as source files are.

stevstrong, 2024-09-14:
What is actually this file good for? I never used that.

rogerclarkmelbourne, 2024-09-15:
Its the COM port driver installer. 

It's derived the example file from 

https://github.com/pbatard/libwdi

with small changes made for the USB PID/VID values originally defined by LeafLabs for their Maple board

I often see false positives from VirusTotal on a variety of files.

If you look at the report from VirusTotal, only 3 of the "antivirus" scanners they use, reported a problem.

Max Secure: "Trojan.Malware.74336512.susgen"    - where the 'susgen' part of the designation means suspicious, but isn't actually a match with a known malware.

SecureAge: "Malicious"  - is meaningless. i.e This scanner has not matched the contents of the file with any known malware

Zillya:   Adware.ExtGPi.Win32.32  - is almost certainly a false positive

If you look at the list of virus scanner which didn't find any problems with the file, there are many very expensive antivirus solutions, including the famous / infamous "CrowdStrike Falcon" , Avast, McAfee, Kaspersky etc etc

There is nothing that can be done about the lower quality malware detectors, which falsely report the file as a virus.
Its very common.  Just search Reddit etc for the false positives listed in the VirusTotal report and you'll find loads of instances where they are false positives

Ultimately its the end user's responsibility to decide whether they want to install a file. 

Its not this projects responsibility to make sure that all malware scanners don't report false positives