rogerclarkmelbourne/Arduino_STM32 #717: Missing implementation of ltoa()

rtek1000, 2019-11-19:
Implementation ref: https://github.com/arduino/ArduinoCore-nRF528x-mbedos/commit/05730711a2ad8748c6e6d1a0f04377e90ca74a0a

Requesting Library, CH376 (USB Host):

`ltoa(ul_size, strNumber, 10);// convert long to string`

https://github.com/djuseeq/Ch376msc/blob/f6c6860ff428c8797b3ab834fb1baf210ea40a65/src/SetGet.cpp#L50

Lib https://github.com/djuseeq/Ch376msc

Arduino library for CH376 file manage control chip
Support read/write files on USB flash drive. The chip support FAT16, FAT32 and FAT12 file system

rtek1000, 2019-11-19:
Note: Arduino_Core_STM32 core compiles without errors

rtek1000, 2019-11-19:
Sketch for test:

https://github.com/djuseeq/Ch376msc/blob/master/examples/basicUsageSPI/basicUsageSPI.ino

stevstrong, 2019-11-19:
You could use itoa (integer to ascii) instead.

dewhisna, 2019-11-19:
`ltoa`, when it's available, is part of `stdlib.h` which is part of `libc` from the compiler tool-chain.  Thus, its presence/absence is compiler dependent and isn't part of the Framework libraries (and so it isn't and shouldn't be part of `Arduino_STM32`).

Both the `itoa` and `ltoa` functions are non-standard, as described in these links [https://stackoverflow.com/questions/190229/where-is-the-itoa-function-in-linux](https://stackoverflow.com/questions/190229/where-is-the-itoa-function-in-linux) and [http://www.cplusplus.com/reference/cstdlib/itoa/](http://www.cplusplus.com/reference/cstdlib/itoa/).  It's not part of ANSI-C and may or may not be in the compiler that you are using.  And it also looks like the code you reference that's trying to use this function has failed to even include `stdlib.h` (i.e. an include what you use bug).

It does look like this compiler actually implements `itoa` though, so as @stevstrong  suggests, you could make a `ltoa` wrapper that will call through to `itoa`, but don't be surprised if on a future compiler environment that that starts breaking too.

You should update the library you are trying to use to make use of `sprintf` or something instead and send them a PR to help them fix it.

rtek1000, 2019-11-19:
It makes no sense to modify the library, if for native Arduino boards no error occurs.

dewhisna, 2019-11-19:
For now it works.  But a future version of gcc for that target will likely drop it and it will break.  As I recall, they are still using a very old version of gcc.  So it **does** make sense to modify the library and use functions that are actually portable across compilers.

stevstrong, 2019-12-26:
> It makes no sense to modify the library, if for native Arduino boards no error occurs.

Well, it does in my opinion.

First, you cannot expect that a whole core should be changed just because one specific library requests the existence of a function, which is normally part of the GCC toolchain.

Second, that function would not even be needed, because the indicated line 50 is active when the number to be converted is less than 1024, which means you can use itoa as a safe replacement, no need at all for long precision. Just try. With this regard, you may post an issue on the original library to be changed. It would be an improvement for Arduino boards as well.

rtek1000, 2019-12-26:
Thank you for your attention. At the moment I am no longer trying to use the Arduino platform. I am using the STM32CubeIDE because USB support is easier to implement. The STM32F407 chip has two USB HOST ports, and I no longer need to use the CH376 or MAX3421. I've been able to get almost everything I need to work, like ILI9341 display, W25Q64 flash memory etc. Thanks.

closed 2019-12-26 by rtek1000