rogerclarkmelbourne/Arduino_STM32 #720: stm32f103ret6 flash reallocation to user application

gsaikiran29, 2019-11-28:
hello,
I am using Arduino_STM32 library. Now I want to Develop an In Application Programming(IAP) like a bootloader. Take Code from SD card and Store in Controller flash at 0x8010000 and then jump to that location and run the loaded new application.

the code took from SD and stored into flash successfully done but after jump function called its printing null like this

13:29:04.933 [RX] - Starts............<CR><LF>

13:29:07.970 [RX] - 20480 uploaded successfully.

<NUL><NUL><NUL><NUL><NUL><NUL>

my Main IAP code is

Header file
`#include <SPI.h>
#include <SD.h>
#include <EEPROM.h>
#include <stdint.h>
#include "libmaple/scb.h"
#include "usb_lib.h"

typedef void (*FuncPtr)(void);
#define SCS_BASE   ((u32)0xE000E000)
#define NVIC_BASE  (SCS_BASE + 0x0100)
#define SCB_BASE   (SCS_BASE + 0x0D00)
#define SCS      0xE000E000
#define NVIC     (SCS+0x100)
#define SCB      (SCS+0xD00)
#define STK      (SCS+0x10)

#define SCB_VTOR (SCB+0x08)
#define STK_CTRL (STK+0x00)

#define RCC   ((u32)0x40021000)
#define RCC_CR      RCC
#define RCC_CFGR    (RCC + 0x04)
#define RCC_CIR     (RCC + 0x08)

#define USB_LP_IRQ  ((u8)0x14)
#define SET_REG(addr,val) do{ *(volatile uint32_t*)(addr)=val;} while(0)
#define GET_REG(addr)     (*(vu32*)(addr))

#define pRCC ((rcc_reg_map *) RCC)

typedef struct {
  u8 NVIC_IRQChannel;
  u8 NVIC_IRQChannelPreemptionPriority;
  u8 NVIC_IRQChannelSubPriority;
  bool NVIC_IRQChannelCmd; /* TRUE for enable */
} NVIC_InitTypeDef;

#define RegBase  (0x40005C00L)
#define CNTR    ((volatile unsigned *)(RegBase + 0x40))
#define _SetCNTR(wRegValue)  (*CNTR   = (u16)wRegValue)
#define _GetCNTR()   ((u16) *CNTR)
#define ISTR    ((volatile unsigned *)(RegBase + 0x44))
#define _SetISTR(wRegValue)  (*ISTR   = (u16)wRegValue)
#define _GetISTR()   ((u16) *ISTR)

void setupFLASH();
void usbDsbISR(void);
void nvicInit(NVIC_InitTypeDef *NVIC_InitStruct);
void nvicDisableInterrupts();
void setMspAndJump(u32 usrAddr);
void systemReset(void);
RESULT usbPowerOff(void);

void setupFLASH()
{
  /* configure the HSI oscillator */
[... 290 lines cut ...]
* Linker script for Generic STM32F103RE boards.
*/
MEMORY
{
ram (rwx) : ORIGIN = 0x20000000, LENGTH = 64K
rom (rx) : ORIGIN = 0x08000000, LENGTH = 512K              //in stm32f103re.ld

/*
* These addresses are where usercode starts when a bootloader is
* present. If no bootloader is present, the user NVIC usually starts
* at the Flash base address, 0x08000000.
*/
#if defined(BOOTLOADER_maple)
#define USER_ADDR_ROM 0x08005000
#else
#if defined(BOOTLOADER_robotis)
#define USER_ADDR_ROM 0x08003000
#else
#define USER_ADDR_ROM 0x08010000           //in boards.h
#endif
#endif
#define USER_ADDR_RAM 0x20000C00

Is this process correct,
please help me where I am doing wrong

stevstrong, 2019-11-28:
Don't expect that some will debug your code, it looks very complicated.
Try to reduce your application to a minimum of code lines which shows your issue.
Try to build up your application step by step.

gsaikiran29, 2019-11-28:
Thank you stevstrong,
my issue is "Jump to User Application" 
`void` setMspAndJump(u32 usrAddr)
{
  typedef void (*funcPtr)(void);
  u32 jumpAddr = *(vu32 *)(usrAddr + 0x04); /* reset ptr in vector table */
  funcPtr usrMain = (funcPtr) jumpAddr;
  SET_REG(SCB_VTOR, (vu32) (usrAddr));
  asm volatile("msr msp, %0"::"g"(*(volatile u32 *)usrAddr));
  usrMain();
}`
I have seen my user application code uploaded successfully or not by using STLink utility.
what care should be taken before Jump function called, and after ?
please show a path to get it done.

rogerclarkmelbourne, 2019-11-28:
I think this is not the appropriate place for this.

Its not an "issue".

Post to a STM32 related forum

closed 2019-11-28 by rogerclarkmelbourne

gsaikiran29, 2019-11-28:
Hello, rogerclark
I know that this is not a stm32duino issue but i cannot find a Right place to ask this question. I thought here i get some suggestions to do this.