rogerclarkmelbourne/Arduino_STM32 #721: Longer delay in /variants/generic_*/wirish/boards_setup.cpp

arpruss, 2019-11-30:
From my experimentation, the ` for(volatile unsigned int i=0;i<512;i++);` delay for the GENERIC_BOOTLOADER setting in /variants/generic_*/wirish/boards_setup.cpp is not long enough for my Raspberry PI 3 to recognize the switch from the bootloader's USB profile to a new USB profile. (I tested this in the context of my USBComposite library, but the issues should be the same in the core.) Replacing 512 with 5120 solves the problem. That's such a slight timing increase, and normally it happens only on startup, that it shouldn't bother anyone else if it were fixed.

rogerclarkmelbourne, 2019-11-30:
I tried this about a year ago.

The 3B is OK but the 3B+ has the problem

However if you lengthen the delay so that it works on the 3B+ the bootloader no longer enumerates on other hardware e..g PCs and possible not on Macs

arpruss, 2019-12-02:
Yeah I have the 3B+.
I wonder if there is a way of getting the timing just right to work on all devices?

rogerclarkmelbourne, 2019-12-02:
Perhaps, but it would be difficult to make it reliable for all devices.

It’s also an issue with the bootloader, because it uses the same timing.

The problem is really a fault on the 3B+ only. So changing everything just because one board has a bug, is probably not a good idea , especially as it’s fundamental to the operation of the code and the bootloader

arpruss, 2019-12-03:
I did some testing on my 3B+ today, and I found that 512 is very close to being right, but it's not fully reliable. At one point in the testing, I was having consistent failure at 508 and consistent success at 509, but reliability probably depends on CPU load and other factors, as in earlier testing 512 wasn't working.

My guess is that a modest increase, e.g., to 600, might fully take care of the 3B+ problem. But I have no idea what it might do to other devices. 

For now, I am running this python script on my 3B+:
`#!/usr/bin/python3
import subprocess
from usb.core import find as finddev
from time import sleep
p = subprocess.Popen(['tail', '-f', '/var/log/messages'], stdout = subprocess.PIPE)
while True:
    line = p.stdout.readline()
    if b'New USB device found, idVendor=1eaf, idProduct=0003' in line:
        sleep(4)
        try:
           dev = finddev(idVendor=0x1eaf, idProduct=3)
           dev.reset()
        except:
           pass`

rogerclarkmelbourne, 2019-12-03:
I think I made it work on the 3B+, with a much longer delay, but when I tried the same delay on my Core i7 PC it didn't work.

Different PC's and Linux and Mac boxes could all behave differently, so even if you find a value that works with you PC as well as the 3B+ I'm not sure it would work on everyone's

You could take a look at https://github.com/juribeparada/MMDVM_HS  as they used to use the Arduino STM32 core and had a similar problem with the 3B+

But they may have only done a fix that worked with their limited usage of their firmware

stevstrong, 2019-12-26:
What is the status of this issue?

arpruss, 2019-12-26:
My testing indicates that a slight increase in the delay--even as little as 550 instead of 512--would greatly improve reliability on the 3B+, but Roger is rightly afraid that it might break things on other hosts.

arpruss, 2019-12-30:
I just looked at the official STM32 core, and they do a 10ms disconnect, which I expect is several orders of magnitude longer than our tight little 512 iteration loop: https://github.com/stm32duino/Arduino_Core_STM32/blob/9ba484ee3a65a6dd775100c632a2d8370b23a4d7/cores/arduino/stm32/usb/usbd_if.c

fpistm, 2019-12-30:
Well, I guess the best option for you is to be able to redefine it.
Don't know if easily doable with this core at sketch level?
Maybe using platform.local.txt

arpruss, 2019-12-30:
Right now I mainly use the USBComposite library, and I can do USBComposite.setDisconnectDelay(n). :-)

rogerclarkmelbourne, 2020-02-05:
Its not possible to make this work with both the RPI3B+ and other machines e.g. Windows and Linux etc.
It a limitation of the hardware and also an annoying "feature" of the RPi3B+

closed 2020-02-05 by rogerclarkmelbourne