rogerclarkmelbourne/Arduino_STM32 #723: FreeRTOS900 hangs unexpectedly

sankaramp, 2019-12-02:
helloo,

Code with freeRTOS900 or pubsubclient individually working fine but when I’m using both together the controller hangs unexpectedly…

libraries used: 
1.  https://github.com/rogerclarkmelbourne/Arduino_STM32/tree/master/STM32F1/libraries/FreeRTOS900
2.   https://github.com/knolleary/pubsubclient

Please help me to find reasons for RTOS hanging?

Thanq,
Sankar

GitMoDu, 2019-12-19:
Smells like [RTOS stack overflow](https://www.freertos.org/Stacks-and-stack-overflow-checking.html) to me. That library is not easy on memory.

stevstrong, 2019-12-19:
@victorpv used to support FreeRTOS, but he is not active anymore, so I think this will not be evaluated soon.

victorpv, 2019-12-19:
As Steve mentioned I'm not very active recently, have little free time, but if you can give more details we may be able to find what's wrong.
Does the controller hangs right away? or runs fine for an amount of time before hanging?
If the later, when it hangs, is it always after the same time or after the same input?
I'm with gitmodu that the most likely cause is a stack overflow, the stack size per task is configurable, have you tried changing that? remember the size is per task, so if you have a single task calling multiple functions, they are still all using the same stack. When not using FreeRTOS your program can use the whole RAM as stack until it hits the heap and corruption happen, there is no limit.

gsaikiran29, 2020-01-31:
Hello,
I am also getting the same issue. I changed the stack size and checked for different values and keep max size up to 40k of 64k. It hangs at different different intervals after 1 day, 2 days also. I can not able to find when it hangs.
how can i utilize this stack and heap?
I have 6 tasks in my code they are all with same priority, some of them have more functions and some are small.
Is that stack size per task is dependent on the size of the task?

victorpv, 2020-01-31:
The stack size is defined per task. They can use a default size or something different than default as long as it fits in the memory set to the RTOS. Check the RTOS documentation for that.
You can try using a debugger probe to find exactly were it hangs.
Lacking a debugger you could try tuning a different pin when each task enters and a different one before the tasks yields, if you have any yields in the tasks. That would help figure out if it always hangs in the same task.
It's really difficult to give advice without more details. If your tasks use any shared peripheral, make sure you use semaphores so 2 different tasks don't try to modify peripheral's registers at the same time.
FreeRTOS also has some functions to track and provide a high water mark for the stack of each task, to show whether any tasks if reaching the bottom of that stack.
You can also try with FreeRTOS821 to see if it makes a difference, although probably not.

gsaikiran29, 2020-01-31:
Thank you @victorpv
Up to now I kept my per task stack size set to 1024. Now I increased that to double for which task is using more functions and structures.
and also I checked from different sources regarding stack and heap
these links may be helpful
(https://www.freertos.org/FreeRTOS_Support_Forum_Archive/January_2007/freertos_About_Stack_and_Heap_1642498.html)
https://www.beningo.com/embedded-basics-7-tips-for-managing-rtos-memory-performance-and-usage/

I will to debug and also check the old 821 library as you said.

victorpv, 2020-01-31:
Same serial print would work but only with Semaphores. That way if a task is using Serial another can't use Serial. Once the first task completes the Serial.print call, can release the semaphore, and then the other task can take it and print.
Semaphores block access to a resource so only 1 task at a time can use it. Otherwise at best you could get in the serial an intermix of both cut at random characters, and at worst get a crash.
Not sure how safe digitalRead() is, but I think works fine.
If any function uses global variables or global peripherals and is called from multiple task, you should use semaphores. If they only use local variables, then should be safe to be used at the same time by different tasks, because the local variables are saved in the stack, and each task has a separate stack, so they don't corrupt each other.
It's a whole lot more complicated than this, this is just a general idea. I recommend you to read the FreeRTOS documentation, and some good documentation on semaphores and when to use them.

gsaikiran29, 2020-02-08:
Thank you @victorpv , as per your suggestions I have removed all Serial prints in my code and increased the stack size per task. It's working somewhat better but when I use gprs(MQTT 2 way communication) with freertos still getting hang issue sometimes.
Is there any heap Issue? could you tell me What is the importance & How to use heap in Freertos?

rogerclarkmelbourne, 2020-02-09:
This isn't really the place to discuss how to use FreeRTOS.

Please consider posting to a FreeRTOS forum or a general STM32 forum like www.stm32duino.com

closed 2020-02-09 by rogerclarkmelbourne