rogerclarkmelbourne/Arduino_STM32 #724: Weird mixture of gpio instructions in SoftWire

profdc9, 2019-12-02:
Everything in this library seems to use the pinmaps for higher speed:

`
void SoftWire::begin(uint8 self_addr) {
    tx_buf_idx = 0;
    tx_buf_overflow = false;
    rx_buf_idx = 0;
    rx_buf_len = 0;
    pinMode(this->scl_pin, OUTPUT_OPEN_DRAIN);
    pinMode(this->sda_pin, OUTPUT_OPEN_DRAIN);
	
	sclDevice = PIN_MAP[this->scl_pin].gpio_device;
	sclBit = PIN_MAP[this->scl_pin].gpio_bit;	
	sdaDevice = PIN_MAP[this->sda_pin].gpio_device;
	sdaBit = PIN_MAP[this->sda_pin].gpio_bit;		
    set_scl(HIGH);
    set_sda(HIGH);
}
`

But if you look at this function

`
uint8 SoftWire::i2c_shift_in() {
    uint8 data = 0;
    set_sda(HIGH);

    int i;
    for (i = 0; i < 8; i++) {
        set_scl(HIGH);
        data |= digitalRead(this->sda_pin) << (7-i);
        set_scl(LOW);
    }

    return data;
}
`
This is just weird. Why isn't the instruction this way?
`
     data |= gpio_read_bit(sdaDevice, sdaBit) ? (1 << (7-i)) : 0;
`

stevstrong, 2019-12-02:
Have you tried your suggestion? Does it bring some speed improvement?

profdc9, 2019-12-02:
I haven't yet tried it, I was just wondering why this combination of
functions was used.  I don't expect Softwire to be fast.  I was more
concerned about code consistency, so that the code has the fewest
dependencies and therefore doesn't break if something else is changed.  It
probably will help a little because it will avoid accessing the pinmap
array every time a bit read occurs but it probably will not be dramatic.

stevstrong, 2019-12-02:
Well, you are right, `digitalRead()` shouldn't be there.
Please try your suggestion and let us know if it works.

stevstrong, 2019-12-19:
@profdc9 
Any update?

stevstrong, 2020-01-17:
Solved by #747 .

closed 2020-01-17 by stevstrong