rogerclarkmelbourne/Arduino_STM32 #725: The F() fuction availability

sutech, 2019-12-02:
Hi!
I am using a STM32f103c based custom board and my main peripherals are UART, SPI, GPIO and Interrupts.
I have problem when trying to use the serial module as a debug tool. although My code works properly if I disable the serial module, when Its enabled, the Processor gets locked randomly. I have read somewhere it might be due to SRAM overflowing which happens when we call serial.print too many times. The propose the F() function in order to save the constants strings in Flash instead of SRAM. after using this function still I have the same problem.
Is there something that I am missing or the F() feature is not available when we use STM32duino??

stevstrong, 2019-12-02:
Your hangup is most probably related to many serial debug prints within interrupt service routines, which should be avoided. When a serial transmission buffer is full, then it will block the software till the next character can be transmitted.

sutech, 2019-12-02:
Yeah, I have put a serial.print in each interrupt routine. 
I used to do the same before, never had faced this problem. 
Thank you for your information.
Is it an Arduino Serial.print  or an ARM Hardware issue?
  So probably would a one character serial.print solve the problem?

zoomx, 2019-12-03:
serial.print in interrupts should be avoided in Arduino too
[](https://forum.arduino.cc/index.php?topic=94459.0)
[](https://arduino.stackexchange.com/questions/46595/sending-serial-data-in-an-interrupt)

sutech, 2019-12-04:
thank you very much
so it seems stm32duino does not utilize the NVIC feature of ARMs.

closed 2019-12-04 by sutech

stevstrong, 2019-12-04:
> so it seems stm32duino does not utilize the NVIC feature of ARMs.

Which feature do you mean?