rogerclarkmelbourne/Arduino_STM32 #728: firmware appears to load but "USB Device Not Recognized"

bill-orange, 2019-12-10:
The firmware appears to load successfully in my bluepill (likely a clone or counterfeit).  The LED blinks after the load.  However, all I can get to appear in device manager is "USB Device Not Recognized", Unknown USB Device (Device Descriptor Request Failed).

How should I proceed?  I am not sure how to troubleshoot this?  Is the problem likely in the firmware side or the USB Driver side?

stevstrong, 2019-12-26:
This is most probably a USB driver issue, please check this how install the drivers:
https://github.com/rogerclarkmelbourne/Arduino_STM32/wiki/Installation

JiriBilek, 2019-12-26:
Or it could be bad soldering issue. 
I got this error on my custom board when the USB connector had bad contact on one pin. You should make sure the USB D- and D+ signals are connected to PA11 and PA12 respectively.

bill-orange, 2019-12-26:
@JiriBilek Good though on the solder connections.  Unfortunately, I checked them and they were fine.

@stevstrong  I repeated the driver installation exactly per the referenced instructions.  It made no difference.

Perhaps I just have a bad board or its fake/clone design deviates for a true device in a significant way.  I only have the one board to play with.

JiriBilek, 2019-12-27:
@bill-orange Really silly question: did you connect BOOT0 jumper back to 0 after flashing the bootloader?

bill-orange, 2019-12-27:
When using serial to burn the bootloader, I did move the jumper back. More recently,  my STLINK arrived from China making the jumper move unnecessary. Good thought. One thing that I have learned providing advice on 3D printing forums is don’t overlook the obvious.

stevstrong, 2019-12-27:
When installing the serial driver, be sure to uninstall any other previous driver.
Anyway, if you do not intend to use USB serial, then you can upload with STLink.
In that case I think we can close this issue, right?

bill-orange, 2019-12-27:
I had uninstalled previous drivers.  I suppose I just have a 'nonconforming' board.  

Let's leave this open for a few weeks to see if anyone else chimes in with me.  It would be good to know if there are a bunch of nonconforming boards showing up out there or if there is some subtle bug with a recent windows update.

closed 2020-02-05 by rogerclarkmelbourne