rogerclarkmelbourne/Arduino_STM32 #731: Maple mini halts after writing in backup RTC registers

hyperion11, 2019-12-13:
Hi! I found this problem on latest framework.
If if write several times in backup register with command setrtccr() or setbkpdrift() using this library
https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/master/STM32F1/libraries/RTClock/examples/RTCAdj/rtadjust.cpp MCU halts on second write. OR! If i write to backup register with command setrtccr() or setbkpdrift() once AND then set RTC with command rt.setTime(tm) MCU halts.
@ag88 are you not found same problem?

ag88, 2019-12-13:
oops, i think i'd need to revise about this as i kind of figured out how it works back then while i'm implementing it. setrtccr() is called by calibratertc(time_t time_now) which is the main entry point when one is setting up the compensation offsets initially.
among the things to look out for is that access to the back up registers requires some sort of initialization codes, this is encapsulated by the bkp_init() function.
```
void setup() {
  /* initialise access to backup registers,
   * this is necessary due to the use of backup registers */
  bkp_init();

  /* adjust rtc */
  adjtime();
}
```

hyperion11, 2019-12-13:
@ag88 
this
  bkp_init();
  adjtime();
already in setup section.
I have crystal with big drift. About 10-20 minutes per month. With your library i can reduce this drift to 1-2 min per month (calibration after 3 days). I want add code to program to change bkpdrift register manually to "fine tuning" it. If clocks still running too fast, increase it slightly, without calibration via serial port. (my device if car computer, fitted in car. It difficult to remove it for experiments).
This behavior occurs if i set time 3 times via serial (command sYYYY-MM-DD HH:MM:SS). And i think it happens after some changes in framework. Because with older (02.19) i not have this problems. But and not try change it several times..

ag88, 2019-12-16:
hi, as i'm kind of busy with other matters, it is unlikely i'd find time to work on this currently.
what i've done prior is that after i set the time using calibratertc() and if it freezes, i simply press reset. and i use the t command showtime() function to check if the relevant rtc and backup registers are updated, they are normally updated. i think the halts are possibly related to writing to the backup registers but i've not had time to figure out the precise cause.
i've not actually made it delta update capable if i remember, you would need to review the codes if you intended to use it that way. it is originally intended to be do an initial synctime() to clock time, let the drifts accumulate and do a calibratertc() and this would compute and save the drift adjustments in the backup registers.
then each time after reset, power up, in
```
void setup() {
  /* initialise access to backup registers,
   * this is necessary due to the use of backup registers */
  bkp_init();

  /* adjust rtc */
  adjtime();
}
```
this would apply the adjustments so that the rtc drift is compensated. normally no further calibration is needed until you want to do a synctime() and calibratertc() cycle again.
and you could read the time off the rtc using the various normal rtc functions.
hope this helps

ag88, 2019-12-16:
my thoughts are that you could still experiment with setrtccr(), the stall may be related to needing write access to the backup registers etc. hence, you may need to review the reference manual rm0008 and codes such as bkp_init() as you work that. i'm not too sure if a call to bkp_init() just prior help. it is likely you can use the hardware adjustments in addition to the software adjustments for 'fine tuning'

rogerclarkmelbourne, 2020-02-09:
Closed because of no activity

closed 2020-02-09 by rogerclarkmelbourne

ag88, 2020-02-14:
ps, i may make a pr for this, i'd put this on my to-do list. i.e. offer an option for 'incremental' adjustments, this would unfortunately use one more backup register, so there'd be 'ifdefs' in my 'example' code

rogerclarkmelbourne, 2020-02-15:
Submit a PR and it will be reviewed

ag88, 2020-02-15:
i added a new PR
https://github.com/rogerclarkmelbourne/Arduino_STM32/pull/753
RTCAdjustment example add incremental calibration 
Incremental calibration allows the user to call calibratertc() repeatedly.
This is done by running synctime() once and you can then call calibratertc(() more than once.

calibratertc() computes the drift adjustments by adding up the cumulative adjustments each time the user call calibratertc(). By doing this, it accounts for the smaller differential errors detected from the elapsed time since the initial synctime() call, and hopefully provide more accurate adjustments over a longer time period.

unfortunately this use additional 2 backup registers (to keep the time of initial time sync), making a total count of registers used 5. i've thought about other means, but those may have a poor convergence. this would probably help those who actually use the codes as a library in a sense it is more self-contained, it is probably not necessary to edit the codes.

so initially call synctime() then afterwards you can call calibratertc() more than once with the new time to sync, for each time it measures the total elapsed time difference from the initial time sync to get the differential error. And add this incremental error to the drift_duration which is basically the 1 in N seconds error with each calibratertc() measurements