rogerclarkmelbourne/Arduino_STM32 #734: STM32F103RCT6 - Support

ETE-Design, 2019-12-18:
Could you Pls. Add this Board? Works perfect om Marlin 2.0 but would like to use MK4Due instead...

stevstrong, 2019-12-19:
I don't think that this board will be added, I don't have such a board so I cannot test it.
The only place I have found the RCT6 version is here: https://web.archive.org/web/20190316171816/https://www.stm32duino.com/viewtopic.php?f=28&t=3837

ETE-Design, 2019-12-19:
maks Robin Lite 1.0 use the chip :-)

victorpv, 2019-12-19:
I have compiled multiple times for RCT6 boards, can't remember if it's in a modified core though.
Have you tried to compare with R series as Steve suggested?

victorpv, 2019-12-27:
The STM32F103RC variant is already in the core, and works fine for me.
Select Generic R series as board,, then RC variant.

@stevstrong I think this one should be closed.

stevstrong, 2019-12-27:
Based on Victor's opinion this should work, so I close this now.

closed 2019-12-27 by stevstrong