rogerclarkmelbourne/Arduino_STM32 #737: STM32F103R8T6

richardmellersh, 2019-12-25:
G'day Roger, I have a CooCox 'Embedded Pi' with the above chip but no matter what I try the Arduino IDE refuses to connect with it - either on my RPi or my iMac. All the resources I have seen seem to point to you as the best source of information. I do hope you can point me in the right direction. Thanks

stevstrong, 2019-12-26:
What exactly have you tried?

It seems that the only programming mode supported by that board is serial, so you have to select "Serial upload" method in Arduino IDE, and follow the instruction from the user manual:

> 3.8.1 ISP mode
In ISPmode, a PC programs the MCU onboard via the serial port (JP7-TX1 and JP7-RX1), refer to section 3.6.1. To  use  this  mode,  you  need  to  set  BOOT0  to  1  (high  level),  and  BOOT1  to  0  (low level) –which has been done on hardware. In this case, you only need to press the BOOT0 button to enter this mode when Embedded Pi is powered on.

richardmellersh, 2019-12-26:
> 07774258369
> 01869 346434

Thanks for your reply.
I’ve set the Ardino IDE as follows;
Generic STM32F103C, Serial, I tried both 72MHz &48MHz,with port-serial
I figured that if it would ‘Get board Info’ then it would work - it didn’t
- the message in the IDE is "the selected serial port does not exist… her
is the 'copied' message:

Arduino: 1.8.10 (Mac OS X), Board: "Generic STM32F103C series, STM32F103C8
(20k RAM. 64k Flash), Serial, 48Mhz (Slow - with USB), Smallest (default)"

Sketch uses 13052 bytes (19%) of program storage space. Maximum is 65536
bytes.
Global variables use 2504 bytes (12%) of dynamic memory, leaving 17976
bytes for local variables. Maximum is 20480 bytes.
stm32flash Arduino_STM32_0.9
Error probing interface "serial_posix"
Cannot handle device "/dev/cu.DT-9663BT-SerialPort-1"

Failed to open port: /dev/cu.DT-9663BT-SerialPort-1
http://github.com/rogerclarkmelbourne/arduino_stm32

Using Parser : Raw BINARY

the selected serial port
 does not exist or your board is not connected

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Even though I did - this time - press the ‘Boot0’ button!

I’m using an iMac with the Arduino IDE, usb serial cable to the usb
connector on the end- the green led ‘ON’ comes on and the other ‘led1’
flashes at about a 1 second cycle. The flashing does not change when press
the ‘boot0’ button. I’m assuming that this led1 is the same as the on on
pin 13 of the Arduino UNO which I have also tried with success.
Thanks again for your reply, just to put my work in context - I’m retired
and have been asked if I will help a youngster out by teaching electronics,
we are at the stage of linking his programming (html) knowledge to the
hardware.

Oh and btw I feel an affinity with all in Oz as I have friends at MAFF in
the Northern Territories and my maternal granfather - a survivor of
Gallipolli - is buried in Payneville.

All the best
Richard
PS: compliments of the season to you and yours

stevstrong, 2019-12-26:
Ok, so you are o MAC.
Unfortunately, I have no experience with MAC.
But if you have an USB-serial adapter, you should see at least the serial port on MAC.
If you do not see it (and this is obviously the case), then you may face a driver issue.
So please install the USB-serial driver first.

For upload you have to set the MCU in the DFU upload state: press the BOOT0 button, press reset, then release reset while keep pressing BOOT0. At this stage the PC should receive some data from the MCU. Then release BOOT0, too.

In the Arduino IDE you have to select "Generic STM32F103R series", because the board has an STM32F103RBT6 chip.

richardmellersh, 2019-12-26:
Ah,ha great thanks I’ll try that right away. Your comments re iMac are
interesting as the Arduino IDE - exactly the same - are also available on
the Pi and the results are the just the same on my Pi 3B & the 3B+.
But lets see what happens with the right ‘magic spell’.
Hmmm I'm on the Pi now to finish this mail but I'm sorry to confess that 7
decades of not being a programmer leads me to much frustration when trying
to identify just which file to download.
I looked back onto github to try to download the right file that would give
me the option of the STM32F103R in the IDE but I just cannot see the right
one - help please?

Thanks
Richard

victorpv, 2019-12-27:
Richard, do you have any bootloader loaded to the MCU? If not, you need to load one.
When using a bootloader, the MCU communication will be trough USB, with a serial port emulation.
Also when selecting the uploade method in Arduino, you will need to select "stm32duino bootloader".

If you want to use a usb-serial converter, then you don't use the stm32f103r usb port. Instead you will use a usb-serial adapter, then connect the TX from the adapter to USART 1 RX in the STM32 mcu, and RX from the adapter to TX in the MCU. Then you can use the rom bootloader in the MCU (that's loaded from the factory, so you don't need to load). That rom bootloader is the one that will require you to set boot0 and boot1 to certain levels and reset the MCU.

There are multiple posts online with all that, it's kind of hard to explain in a github issue post, so I advice you to better search on google for some instructions on that.

Just remember: MCU USB port requires bootloader loaded first to the board, then select "stm32duino bootloader" as upload method in Arduino, and I believe in OSX you need a driver.
MCU serial port does not require a bootloader (is in ROM), but in that case you need a usb-serial adapter and connect that to the pins for USART 1 TX & RX in the board, then select "serial" as upload method in Arduino.

richardmellersh, 2019-12-31:
Whoa, “boot loader”? The last time I played with one of thos was 1975 on a
Data General ‘Nova 1200’, flicking switches to load a punch tape.
I have made a discovery though; the board descriptor that does work is from
Dan.Drown.org from a suite of STM files - I had forgotten that, on the iMac
I had to put a line into the Arduino IDE that pointed to Dan.Drown’s web
site under preferences/Additional Boards… Iremembered this when I went back
to the iMac but when I tried to insert the same line in the Pi version of
Arduino IDE a message appeared in the messegaes box that the web site ould
not be reached??? After more investigation, it would seem the Dan Drown?
has not released a version for the Pi and so the board driver that works
just fine for the imac (and, I guess Windows) isnt available to run the
board for which it was built ie. the Embeded Pi - strange - I searched for
a way to contact the Dan Drown but to no avail.

I am now at a loss and will have to run the Embeded Pi on my imac or
MacBook I guess.

Thanks for your help and I wish you a very happy & successful new Year

Cheers
Richard  Mellersh

closed 2020-02-05 by rogerclarkmelbourne