rogerclarkmelbourne/Arduino_STM32 #739: Bluepill SHT20 sensor interface issue

abhaysbharadwaj, 2019-12-27:
Hello,
I am trying to interface the SHT20 sensor from DFRobot: https://wiki.dfrobot.com/SHT20_I2C_Temperature_&_Humidity_Sensor__Waterproof_Probe__SKU__SEN0227
The given example code and library is working well with Arduino Nano and ESP12E NodeMCU board.
But when I use with Bluepill board, the program outputs "998" for both temp and humidity, which according to the library .h file means "i2c timeout".
I am unable to understand how to fix this. Any help on this will be much appreciated.

dewhisna, 2019-12-27:
Make sure you are on the current master of this `Arduino_STM32` repo as there was a fix in #704 that addressed a similar issue.  Also, verify that you have proper 4.7K ohm pull-up resistors on both the SDA and SCL lines.  Some other boards have their own pull-ups but your blue-pill board probably does not.  So if you haven't added these pull-ups, it probably won't work.  Looking at the article you linked, they mention "a 10K pull-up resistor" built into the sensor, which doesn't make much sense, as there would have to be at least two pull-up resistors -- one on the SDA and one on the SCL -- not just one.  And 10K is a bit on the weak side.  4.7K resistors would be much better.

So 1) Make sure you are on the latest code of the master branch, 2) Add some 4.7K pull-up resistors on both the SDA and SCL to 3.3v, and 3) double-check your connections that you don't have SDA and SCL swapped or something.  If all else fails, perhaps try an [I2CDriver](https://i2cdriver.com/) to debug it.  It's an extremely useful tool.

stevstrong, 2019-12-27:
@abhaysbharadwaj 
beside what @dewhisna indicated, please also provide an example which shows the issue.

abhaysbharadwaj, 2019-12-29:
Hi,
Thank you for the quick revert. I am currently traveling. I'll be back home
tomorrow. I will do what you have suggested and revert back.

abhaysbharadwaj, 2019-12-30:
@dewhisna I can partially confirm that the issue was with the STM32duino core.

So I had an old laptop which I was using from the past 3 years. It was not in great shape but my environment was configured perfectly. The same sensor and board with the same example code are working perfectly with it.

The new laptop that I got, I downloaded and installed things around October 10th 2019. The Issue I mentioned above is coming up in that. I will update the core in my new system and revert back here by end of the day, but as a quick revert, the sensor, library and code is working perfectly with the old core running on my old system.

Let this issue be open for now. I will confirm again soon.

P.s. the sensor has two 4.7K resistors on the sensor PCB on SDA and SCL lines.
This is the example image:
[image]

[image]

rogerclarkmelbourne, 2019-12-30:
The core may have changed from software to hardware I2C between those versions.

Its most likely a pullup resistor or hardware speed problem.

You could try STM's official core, or post to www.stm32duino.com but just because your specific hardware is not now working with this core does not mean the core has a bug

dewhisna, 2019-12-30:
@abhaysbharadwaj  you mention that it wasn't working with the code from "around October 10th".  That was very close to the time when PR #706 landed and it's very possible you were using the code just prior to that, which had an issue with 2-byte reads, which was causing a similar problem on the BME280 sensor.  Therefore, it's imperative that you test it with the latest code so we aren't chasing problems that are already fixed.

When you run your tests and reply with the results, please include the Git hash for the code you are using so that I will know I'm looking at the correct code and the same code you are testing.

If it persists with the latest code and all of your pull-up resistors and connections are good, we will need your help in diagnosing it, since I don't think anyone else has that sensor to do any testing with (I know I don't).  You will need to use something like the aforementioned [I2CDriver board](https://i2cdriver.com/), or other similar way of looking at the waveforms on the bus, to figure out exactly what the problem is.

abhaysbharadwaj, 2019-12-30:
Hello,
in short, the sensor and code is working fine with the latest stm32Duino core version "2019.12.28".

I had got the issue in version "2019.10.9" in my new system. I updated the same to the latest version and it is working fine.
Also, my old system is using version "2019.2.3" and hence this issue was not observed there.

you can go ahead and close the issue. I apologise if any inconvenience was caused and also, thanks a bunch for the quick revert and support!

abhaysbharadwaj, 2019-12-30:
Just an addon. if someone wanted to try interfacing with the sensor. These are the specifications:
1. Board used: STM32F103C8T6 based bluepill.
2. Sensor: SHT20 from DFRobot (https://www.dfrobot.com/wiki/index.php/SHT20_I2C_Temperature_%26_Humidity_Sensor_(Waterproof_Probe)_SKU:_SEN0227)
Sensor has onboard 4.7K resistors on SCL and SDA line. VCC is 3.3V.
3. Arduino IDE version: 1.8.10
4. STM32Duino core version: 2019.12.28

attaching a .zip file with the code (.ino) and library (.h and .cpp) that was used for testing. note: the code and library are same as that available on DFrobot's wiki page.

[DFRobot_SHT20_sensoronly.zip](https://github.com/rogerclarkmelbourne/Arduino_STM32/files/4010249/DFRobot_SHT20_sensoronly.zip)

closed 2019-12-30 by abhaysbharadwaj

dewhisna, 2019-12-30:
@abhaysbharadwaj -- thanks for the update.  I think the code in PR #706 fixed your problem and you just happened to be on a code snapshot from before that in your previous tests.  So in essence, your issue was a duplicate of #704.  I figured that was the case.  Glad to hear that it's working for you.

In the future, when reporting core versions, don't use just the date, as that's somewhat ambiguous and difficult to track, especially since commits can be dated and sitting on another branch long before they get merged to master.  And if there are commits on the date specified, are you referring to the code-point before those commits or after them?  Instead, use the git commit hash so that it references a very specific code-point.  And that applies to every git repo, not just this one.

abhaysbharadwaj, 2019-12-30:
@dewhisna  -- yes, as mentioned by you the PR #706 must have been the thing that fixed the problem as I was not even getting any response when using the i2c scanner code.

And I will try to follow the core versioning mentioned by you. I am a complete hardware dev guy and firmware dev is something I am learning. Git is like a pandora's box to me and trying to figure things as I go about.
The core versioning is what I saw in the Arduino IDE's "Board manager" as below:
[image]

I understand your point of mentioning the exact commit so tracing and identifying the bugs/issues will be easier and I will try to follow that!
Again, thanks a bunch for the support! cheers!