rogerclarkmelbourne/Arduino_STM32 #740: delayMicroseconds() half of correct time

arpruss, 2019-12-30:
On my blue pill, delayMicroseconds() consistently does only half of the required delay; delay() works fine (or else both millis() and delay() are wrong :-) ). The following code outputs 5012 or 5013:

```void setup() {
  Serial.begin();
}

void loop() {
  uint32 t0 = millis();
  for(int j=0;j<1000;j++) {
    delayMicroseconds(10000);
  }
  uint32 t = millis() - t0;
  Serial.println(t);
}
```

arpruss, 2019-12-30:
It's a little better at 48mhz instead of the default 72mhz: instead of the delay being half, it's 2/3.

GitMoDu, 2019-12-30:
Are you changing the clock frequency after boot? What clocks are you using? 
millis() and micros() rely on F_CPU for a correct calculation, if that's not properly defined in the Core, then it will fail.

arpruss, 2019-12-30:
millis() and delay() work just fine. It's delayMicroseconds() that doesn't. I just verified the delayMicroseconds() timing with a stopwatch, and it was pretty much exactly what millis() said it was (i.e., half of the correct delay at the default 72mhz and 2/3 at 48mhz).

I am setting the clock rate with Tools | CPU Speed.

arpruss, 2019-12-30:
An interesting question regarding fixing this will be what to do with the places in the core and libraries that rely on delayMicroseconds() to do delays. After delayMicroseconds() is fixed, should we double the value in all the calls that previously seemed to work?

GitMoDu, 2019-12-30:
Changing the CPU frequency to 48 MHz, changes F_CPU to the correct value (48 000 000), which is used in micros() and delayMicroseconds(). I don't see nothing wrong in that regard.

GitMoDu, 2019-12-30:
Something is definitely fishy. I was able to reliably replicate with this test code I ran.

Sanity checks first:
You're using Roger's core with Maple Mini as a device, selecting the CPU speed like this, correct?

[image]

```
void setup() {

  Serial.begin(9600);

  // Wait for serial to be ready.
  while (!Serial)
      ;

  Serial.print(F_CPU/1000000);
  Serial.println(F(" MHz"));
}

void loop() {
 uint32 t0 = millis();
  for(int j=0;j<1000;j++) {
    delayMicroseconds(10000);
  }
  uint32 t = millis() - t0;

  Serial.println(t);
  while (true)
    ;
}
```

Results:

> 72 MHz
> 11690

```
48 MHz
6686
```

arpruss, 2019-12-30:
The timing of delayMicroseconds() depends on STM32_DELAY_US_MULT in:
https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/9909faf5ce152574d32a7d93632b86ba802507bc/STM32F1/system/libmaple/stm32f1/include/series/stm32.h

Currently, on the 103C6, 103C8 and 103CB, this is set to (F_CPU / 6000000L), which is 12 at 72mhz. My experimentation suggests that for 72mhz this should be (F_CPU / 3000000L), i.e., 24. But if we do that, it'll be off at 48mhz: the speed doesn't scale linearly with the clock rate. At 48mhz, what we need is 12. At least on the unit I've got my hands on.

Interestingly, I also tested a fake C8 (which I've assumed is a rebranded C6, but I don't know), and it's more than twice as slow as the real C8.

arpruss, 2019-12-30:
No, my device wass "Generic STM32F103C series" for the C8 tests (and "Generic STM32F103C6 / Fake STM32F103C8" for the C6 tests). No difference with Maple Mini setting, though.

But things are getting stranger. I just pulled another "C8" off my shelf. While the one I was testing before gave me 5012 as output, this was one gave me something very close to 10000, the correct value.

arpruss, 2019-12-30:
Here's a suggestion: do a calibration at system startup instead of hardcoding STM32_DELAY_US_MULT. The following takes only 10-20ms to run, depending on the board speed, and would let us eliminate all the messy hardcoded delay times for different devices. 

```uint32 getDelayUSMult() {
  uint32 t = micros();
  my_delay_iterations(100ul);
  uint32 t1 = micros() - t;
  t = micros();
  my_delay_iterations(100100ul);
  uint32 t2 = micros() - t;
  t2 = t2 - t1; // this measures the difference between 100100 and 100 iterations, thereby avoiding overhead
  return (100000ul + t2 / 2) / t2;
}
```

GitMoDu, 2019-12-30:
Doesn't make sense to me, those should be compile time constants, something else must be broken.

dewhisna, 2019-12-30:
I keep hearing mention of the fake C8 processors in this thread (i.e. the C6).  And what about the supposed "real C8" mentioned?  Where were they purchased?  Could that be what's "broken"?  Why would anything about a fake processor meet spec, including the PLL dividers for the processor execution clock?  The real question is how well do these timing loops work on genuine processors from ST purchased through reliable distributors?

Though one thing did just occur to me.  When you guys are switching the CPU clock settings, did you do a complete clean build and delete all previously compiled object units and artifacts?  If not, is it possible that something was inlined with a previous setting and you are getting a mixed build where some functions got updated and others not?  The build environment may not have the dependencies correct to rebuild everything like it should when the speed is changed via the IDE.  So do make sure that you're building and testing what you think you are.  And then perhaps look closer at the chip origins, since you are seeing variations between supposedly identical chips.

rogerclarkmelbourne, 2019-12-30:
Some cheap BluePill boards seem to have relabelled chips on them, and people post bugs to this repo, because they are not aware that they have been scammed, and think they have a genuine STM32.

There are multiple scams going on.

Lesser versions of the STM32F103 are getting relabelled, but there is also a clone chip called a CS32 which is also being relabelled as a STM32.

It’s also possible that the GD32 is being sold as a STM32, but this seems far less common, as normally the GD32 is more expensive to buy than the STM32.

Strangely, despite buying loads of BluePill boards over the last few months, I have never seen want of these fake devices

arpruss, 2019-12-30:
The board where delayMicroseconds() works correctly is a black-colored board that has the layout of a Blue Pill (same wrong USB resistor, etc.). I think I got it from the WAVGAT store. I am guessing that's genuine, both because of where I got it and because of delayMicroseconds() working fine.

The Blue Pill where delayMicroseconds() runs twice as fast was bought from a random ebay dealer.

I just ran the Dhrystone benchmark.

Wavgat(?) black board: 63525.80
Mystery blue pill: 72549.93

So my mystery blue pill benchmarks at 14% faster than the one that I think is genuine. (And twice as fast at the very small loop of the delay_us() code.)

Anyway, I guess all this means there is no bug in the core. But I think my suggestion that we spend a few milliseconds doing a benchmark may be worth thinking about. Of course, one can also just do that in one's own sketches and libraries.

arpruss, 2019-12-31:
Oddly,  I re-ran the tests on the fake C8 that I think may be a C6, and everything works fine now on it. But earlier today it was slower. Weird indeed!

GitMoDu, 2019-12-31:
@rogerclarkmelbourne I only tried with a BluePill with the new Bootloader. Wondering it that makes any difference.

arpruss, 2020-01-01:
Here's something weird. delay_us() uses the following loop code:
```
    mov r0, %[us]
1: subs r0, #1 
    bhi 1b
```

My mystery blue pill executes loops like this twice as fast as my other boards. 

However, if I insert a `nop` inside the loop (either before or after the `subs` line), the speed on my other boards doesn't change at all. However, the speed of the loop on the mystery blue pill then goes down to the same speed as on all the other boards. 

That means that a simple fix for the 103c8 would be to insert a `nop` in delay_us() and it would equalize the delay between the boards. But of course the `nop` might change the timing on other devices. I suppose this has something to do with caching or pipelining.

victorpv, 2020-01-14:
In the old forum there was once a thread about using the cycles counter register in the MCU, DWT or something like that was called, I can't remember. I believe either the due or the teensy use that register.
It was not implemented due to some side discussions about debugging and that register, but I think it may still be the best way to have accurate micros in all conditions.

arpruss, 2020-01-14:
Yeah, I use DWT for precise timing, but it requires turning on the debug mode.

stevstrong, 2020-01-14:
As long as only one MCU shows this effect, this may be special case, I don't see any need to make some changes which could have negative impact on all other devices.

closed 2020-01-14 by stevstrong