rogerclarkmelbourne/Arduino_STM32 #741: 10 seconds are actually taking 15-16 seconds, RTClock library, BluePill-RTClock-test.ino example, STM32F103C8T6 board

JazzTp, 2019-12-31:
Library: **RTClock**
Example: **BluePill-RTClock-test.ino**

( ./hardware/Arduino_STM32-master_2019-12-30/STM32F1/libraries/RTClock/examples/BluePill-RTClock-test/BluePill-RTClock-test.ino )

**A 10 seconds time lapse on the Serial Monitor is actually taking about 15-16 seconds (manual chronometer).**

IDE/OS: **Arduino IDE 1.8.10 on Ubuntu Linux 18.04.3**
Board: **STM32F103C8T6**

I'm using Roger Clark's core https://github.com/rogerclarkmelbourne/Arduino_STM32
(I've just updated it, to check if the same occurred with the latest version).

**IDE settings:**
Board "Generic STM32F103C series"
Variant "STM32F103C8 (20k RAM, 64k Flash)"
Upload method "Serial"
CPU Speed(MHz) "72 Mhz (Normal)"
Optimize "Smallest(default)" (other options: Fast, Faster, Fastest, Debug)
Port "/dev/ttyUSB0"
Programmer "USBasp"

I have quite a few other sketches open which use quite a few other libraries, and they are getting compiled and run correctly, except a couple, the ones related to the RTClock of the STM32, [one is failing to compile](https://github.com/ZulNs/STM32F1_RTC/issues/2), and with [another one](https://github.com/cbm80amiga/N5110_STM32_RTC/issues/1) a time lapse of 10 seconds on the Serial Monitor is actually taking 20 seconds.

I guess this core has undergone RTC-related changes since the authors tested those sketches.

stevstrong, 2019-12-31:
Actually, RTC was not touched for a while.
We had issues of deviation in RTC timing which were solved by removing the PC14/15 pins from the board.
It would be nice if you could post in the [forum](https://www.stm32duino.com) before being sure that it is really a core issue.

JazzTp, 2019-12-31:
@stevstrong 

> It would be nice if you could post in the [forum](https://www.stm32duino.com) before being sure that it is really a core issue.

Oh **thank you**, forum **bookmarked**.

Please notice, however, that **the first sketch I mentioned** in my report is **part of this repository**, despite not being part of the core itself (which is why I had posted the full path to the sketch inside the core tree):

`./Arduino_STM32-master/STM32F1/libraries/RTClock/examples/BluePill-RTClock-test/BluePill-RTClock-test.ino`

 (I've just triple checked, after downloading the core again.)

I'll check in the forum if there is some open thread about the RTC examples.

JazzTp, 2020-01-08:
Closing, THANK YOU, solved as reported here: https://github.com/ZulNs/STM32F1_RTC/issues/3#issuecomment-571840963

closed 2020-01-08 by JazzTp