rogerclarkmelbourne/Arduino_STM32 #743: driver installation failure

kartikbathla, 2020-01-03:
[image]
[image]

I am trying to install the drivers for stm32f103c8t6 to use it with arduino IDE. I have set up the IDE with adding the board from http://dan.drown.org/stm32duino/package_STM32duino_index.json .  After unzipping this zip file in the hardware folder in the arduino folder, I ran the driver installer it doesn't seem to have worked properly.

stevstrong, 2020-01-03:
Have you executed the BAT files with administrator rights?

kartikbathla, 2020-01-04:
yes I did, actually it turns out the problem wasn't with this driver but the driver for my USB to TTL. Apparently, the USB to TTL serial convertor that I got with the MCU had a fake PL2303HX chip which are blocked by the latest driver. So I had to switch to a previous driver to get it work for now.  Soon gonna buy one with an original chip.

closed 2020-01-04 by kartikbathla