rogerclarkmelbourne/Arduino_STM32 #745: Serial communication over USB - UNSTABLE

renno-bih, 2020-01-06:
Hello,
I'm using Blue Pill board (1.5K pull-up on PA12) for my project where I implemented very simple polling mechanism. I'm sending command using USB serial every 100ms and listening for reply. 
It is working perfectly but only for some time. After 50-60 minutes Blue Pill freezes and I need to restart it to make it working again. 

Any idea what can cause this issue?

renno-bih, 2020-01-09:
UPDATE: 
Looks like this problem is related to **ST-LINK** programmer. Blue Pill was connected to ST-LINK (ST-LINK part of STM32F3Discovery board) and after disconnecting it from ST-LINK everything is stable. 

No idea why ST-LINK causes Blue Pill to freeze :/

JiriBilek, 2020-01-09:
When connecting both USB and STLink interfaces you have to disconnect 3v3 on STLink otherwise you are in danger of frying the board. Only one power supply can be used.

closed 2020-01-10 by stevstrong