rogerclarkmelbourne/Arduino_STM32 #746: Liquid Crystal Library id buggy!

gunpal5, 2020-01-06:
Hello,

I interfaced a LCD to STM32F103C8T6 board and no matter what I do to circuit it showed garbage text. I kept changing the connection to different ports and pins. but it still remain the same. 

I literally spent 24 hours on this. The LiquidCrystal Library for STM32 published to arduinoIDE is buggy. It shows garbage values.

By using this library https://github.com/SayidHosseini/STM32LiquidCrystal fixed the issue. Please update this in the project, so that other people don't face the same issue.

Thanks

GitMoDu, 2020-01-07:
The repository you linked has a viral license, which is not a great option to include in an open-source , non-viral, core.

rogerclarkmelbourne, 2020-02-05:
Please submit a PR with the fix

closed 2020-02-05 by rogerclarkmelbourne