rogerclarkmelbourne/Arduino_STM32 #748: WireBase::end() not defined

arpruss, 2020-01-20:
Compile:

```
SoftWire* wire;

void setup() {
  wire = new SoftWire(1, 2, SOFT_STANDARD);
}
```

Get:
```
C:\Users\[USER]\AppData\Local\Temp\arduino_build_641112\libraries\Wire\utility\WireBase.cpp.o:(.rodata._ZTV8WireBase+0x14): warning: undefined reference to `WireBase::end()'
```

arpruss, 2020-01-20:
I now know it's a bad idea to use heap (I was using an old library I wrote before I knew that). Here's an example that generates the same error without heap:
```
#include <SoftWire.h>

class Test {
  SoftWire wire;
public:
  Test(int x,int y) : wire(x,y,0) {};
};

Test test(1,2);

void setup() {
}
```

GitMoDu, 2020-01-21:
Related: https://github.com/rogerclarkmelbourne/Arduino_STM32/issues/730
I think you'll get this warning with any implementation Wirebase withtou ::end()

stevstrong, 2020-01-21:
Please try to replace [line 84](https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/master/STM32F1/libraries/Wire/utility/WireBase.h#L84) with this:
`virtual void end() {}`
and let me know the result.

GitMoDu, 2020-01-23:
The warning goes away on my side, since we're now providing default implementation of the virtual method.

stevstrong, 2020-01-23:
Fixed with https://github.com/rogerclarkmelbourne/Arduino_STM32/commit/d8c83f02e520cafead41b009700ce23961aa958e.

closed 2020-01-23 by stevstrong