rogerclarkmelbourne/Arduino_STM32 #749: STM32 and Excel

boromyr, 2020-02-05:
Hi, I currently use this library with Excel to communicate with Arduino: http://www.robertovalgolio.com/sistemi-programmi/arduino-excel
I tried to use it with a Blue Pill version STM32F103C8, and Excel receives the data very well and prints it, but when I want to send the data to STM32 through the .get (...) function it freezes everything, Excel no longer receives the data, and the PC13 LED starts to flash quickly with a sinuous trend.
I managed to isolate the problem to the strcpy function in the rExcel.cpp library lines: 299,301,302,307.
Anyone know why this problem? Is it a compatibility issue with STM32 libraries?
If anyone could do a test with the .get () function I would really appreciate it!

rogerclarkmelbourne, 2020-02-05:
Issues are not a support forum. 

Please try posting to www.stm32duino.com or www.stm32world.com

closed 2020-02-05 by rogerclarkmelbourne

boromyr, 2020-02-06:
I'm sorry