rogerclarkmelbourne/Arduino_STM32 #750: ASSERTs increase code size

me21, 2020-02-06:
There are ASSERTs placed in the code here and there, for example: https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/683e5a371e14fd0ffb72c183599069655a79aec5/STM32F1/cores/maple/HardwareTimer.cpp#L68

They get compiled into the binary by default, increasing its size. Wouldn't be better to turn off debugging by default? It can be done here: https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/683e5a371e14fd0ffb72c183599069655a79aec5/STM32F1/system/libmaple/include/libmaple/util.h#L84

rogerclarkmelbourne, 2020-02-09:
The Arduino system is not intended as an optional production build system.

You can change this setting in your local copy.

closed 2020-02-09 by rogerclarkmelbourne

me21, 2020-02-09:
At least it should be documented somewhere as an easy way to reduce code size...

me21, 2020-02-09:
Ok, I'm sorry, I've got an impression that this change was introduced somewhere during last year. I remember the binary size of my project going up after updating Arduino framework, and when I found strings with paths to project files inside my binary, I thought "that's why".

It is true that Arduino system is not easily tuned, but since Arduino framework is used with other build systems, e.g. PlatformIO (as I do), I guess I can try to talk them into changing default build settings.