rogerclarkmelbourne/Arduino_STM32 #754: Wrong SPI transfer16() function on stm32f4

fa1ke5, 2020-02-16:
In SPI.cpp lib for stm32f4 uncorrect function transfer16(uint16_t wr_data)
`uint16_t SPIClass::transfer16(uint16_t wr_data) const
{
	spi_dev * spi_d = _currentSetting->spi_d;
	spi_rx_reg(spi_d); // read any previous data
	spi_tx_reg(spi_d, wr_data); // "2. Write the first data item to be transmitted into the SPI_DR register (this clears the TXE flag)."
	while (spi_is_tx_empty(spi_d) == 0); // "5. Wait until TXE=1 ..."
	while (spi_is_busy(spi_d) != 0); // "... and then wait until BSY=0 before disabling the SPI."
	return (uint16)spi_rx_reg(spi_d); // "... and read the last received data."
}`
its make only one transaction and work like transfer8
dont work correctly XPT2046 lib and mach more.
this code work better :)
`uint16_t SPIClass::transfer16(uint16_t data) const
{
    // Modified by stevestrong: write & read two consecutive bytes in 8 bit mode (DFF=0)
    // This is more effective than two distinct byte transfers
    spi_dev * spi_d = _currentSetting->spi_d;
    spi_rx_reg(spi_d);                   // read any previous data
    spi_tx_reg(spi_d, data>>8);          // write high byte

	while (spi_is_tx_empty(spi_d) == 0); // wait until TXE=1
    while (spi_is_busy(spi_d) != 0); // wait until BSY=0
    uint16_t ret = spi_rx_reg(spi_d)<<8; // read and shift high byte
    spi_tx_reg(spi_d, data);             // write low byte

	while (spi_is_tx_empty(spi_d) == 0); // wait until TXE=1
    while (spi_is_busy(spi_d) != 0); // wait until BSY=0
    ret += spi_rx_reg(spi_d);            // read low byte
    return ret;
}`

stevstrong, 2020-02-16:
This function is originally designed to make one 16 bit transfer, and not 2 transfers in 8 bit mode, as your version.

stevstrong, 2020-02-16:
If you use the STM32F4 family MCU you may be better served by my repository.
There, yo can find a variety of SPI transfer function versions, e.g.: https://github.com/stevstrong/Arduino_STM32/blob/master/STM32F4/libraries/SPI/src/SPI.h#L285-L289

fa1ke5, 2020-02-16:
ooooh crazy opensource......
one 16bit transfer? but datalogger say me one 8bit transfer.

fa1ke5, 2020-02-16:
> If you use the STM32F4 family MCU you may be better served by [my repository](https://github.com/stevstrong/Arduino_STM32).
> There, yo can find a variety of SPI transfer function versions, e.g.: https://github.com/stevstrong/Arduino_STM32/blob/master/STM32F4/libraries/SPI/src/SPI.h#L285-L289

thanks!

rogerclarkmelbourne, 2020-02-16:
@stevstrong 

You could raise a PR on your F4 repo to merge it into my repo.

I don't think many people use my F4 core as its not actively supported.

fa1ke5, 2020-02-16:
Thanks Roger, i think this board have nice perspectives like bluepill but more and more powerful.
https://os.mbed.com/users/hudakz/code/STM32F407VET6_Hello/shortlog/
some lightly changings and im taking ultra fast fsmc display, touch and match more gpios

stevstrong, 2020-12-13:
The function in question works as expected (transfers 16 bit of data) only when the data size is set to 16 bit.
For that, the user should call `SPI.setDataSize(DATA_SIZE_16BIT);` before performing 16 bit access.
Per default the data size is set to 8 bit access, but it can also be set with `SPI.setDataSize(DATA_SIZE_8BIT);`
To transfer more than one byte in 8 bit mode the user should store the data in an array and use the function `SPI.transfer(buffer, nr_bytes);`

closed 2020-12-13 by stevstrong