rogerclarkmelbourne/Arduino_STM32 #756: about PWM pin map in generic_stm32f103v

louwangzhiyuY, 2020-02-22:
Hi Roger,

i am using  arduino generic_stm32f103v to control fan speed. I want to use PB3 to out PWM to control fan speed, but i found the pin cannot control fan speed, but PA0, PA1, PA2, PA3 is ok. this mean PB3 cannot as PWM pin?  I check PIN_MAP, PB3 don't contain timer device.  do you know how to map PB3 as a PWM pin?

	{&gpiob,   NULL, NULL,  3, 0, ADCx}, /* PB3  */ //JTDO, SPI3_SCK / I2S3_CK/

rogerclarkmelbourne, 2020-02-22:
Not all pins are associated with a Timer and hence can't be used for PWM.

See the STM data sheet for the chip you are using

closed 2020-02-22 by rogerclarkmelbourne

louwangzhiyuY, 2020-02-22:
Hi Roger,

I sure PB3 can be used for PWM from datasheet and it is associated with Timer2 CH2.  can I change the PWM map in STM32 arduino?
[image]

rogerclarkmelbourne, 2020-02-22:
That pin looks like it may be a SWD pin.

Did you disable SWD debugging in the code.

If you don't you can't use that pin at all

louwangzhiyuY, 2020-02-22:
Hi roger,

i don't disable SWD debug,  even if I don't know how to disable it.  so I cannot use PB3 as PWM pin?

rogerclarkmelbourne, 2020-02-22:
Issues are  not a support forum, they are for bug reports.

Please post to one of the user groups, e.g. www.stm32duino.com or www.stm32world.com