rogerclarkmelbourne/Arduino_STM32 #758: stm32f103 frequency counter need some help....

nikjayswal, 2020-02-24:
getting wrong values in frequency counter. the input frequency is 947.1Khz but showing wrong value about 500khz in serial output is the code is wrong ?

Arduino Code:

int pin7 = PA7;
unsigned long pulseHigh,pulseLow,pulseTotal;
float frequency,DutyCycle;
boolean fre;

void Freq_Counter(){
    pinMode(pin7, INPUT);
    pulseHigh = pulseInLong(pin7, HIGH);
    pulseLow = pulseInLong(pin7, LOW);
    pulseTotal = pulseHigh + pulseLow;  //Time period of the pulse in microseconds
    if(pulseTotal < 1){
      frequency = 0;
      DutyCycle = 0;
    }else{
    frequency = 1000000/pulseTotal; //Frequency in Hz
    DutyCycle = (pulseHigh / pulseTotal)*100;
    }
    
    //Serial.println(frequency);
   // delay(127);
  }
  

  void setup(){
    pinMode(pin7,INPUT);
    Serial.begin(115200);
    
  }

  void loop(){

    Freq_Counter();
    Serial.print(frequency);
    Serial.print(" Hz");
    Serial.print("| Duty:");
    Serial.println(DutyCycle);
    
  }

stevstrong, 2020-02-24:
This does not seem to be an issue of the core.
Please post such inquiries under: www.stm32duino.com

It seems that you are at the limit of the processor capabilities.
And, with 500kHz you are under the 10 microseconds limit, thus out of specification as described here:
https://www.arduino.cc/en/Reference.PulseInLong

> The timing of this function has been determined empirically and will probably show errors in shorter pulses. Works on pulses from 10 microseconds to 3 minutes in length.

closed 2020-02-24 by stevstrong

nikjayswal, 2020-02-26:
so how can i measure frequency greater than 100KHz without pulseIn/pulseInLong ?

stevstrong, 2020-02-26:
Again, this does not seem to be an issue of the core.
Please use this forum link (was it you who started the discussion there?): 
https://www.stm32duino.com/viewtopic.php?f=42&t=182

nikjayswal, 2020-02-26:
Yes. And sorry i forgot to check this form. Due to no notification on email.