rogerclarkmelbourne/Arduino_STM32 #759: USB_DISC pin is not generic.

usk-johnny-s, 2020-02-25:
USB_DISC pin is maple specific, not generic.
In generic board configuration, should not determine USB_DISC pin.

My guess is follows. It can use with compiler option like a "-DBOARD_USB_DISC_DEV=GPIOC -DBOARD_USB_DISC_BIT=12".
> #ifndef BOARD_USB_DISC_DEV
> #define BOARD_USB_DISC_DEV      NULL
> #endif
> #ifndef BOARD_USB_DISC_BIT
> #define BOARD_USB_DISC_BIT      NULL
> #endif

https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/0383384f4cc91aadee1501e11bf08dd3e62d9fb9/STM32F1/variants/generic_stm32f103r/board/board.h#L102-L103

https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/0383384f4cc91aadee1501e11bf08dd3e62d9fb9/STM32F1/variants/generic_stm32f103t/board/board.h#L66-L67

https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/0383384f4cc91aadee1501e11bf08dd3e62d9fb9/STM32F1/variants/generic_stm32f103v/board/board.h#L110-L111

https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/0383384f4cc91aadee1501e11bf08dd3e62d9fb9/STM32F1/variants/generic_stm32f103z/board/board.h#L102-L103

and https://www.st.com/en/evaluation-tools/stm32vldiscovery.html did not have USB_DISC pin.
https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/0383384f4cc91aadee1501e11bf08dd3e62d9fb9/STM32F1/variants/STM32VLD/board/board.h#L82-L83

dewhisna, 2020-02-25:
I've had issues with this too, where I was using those GPIO pins for other purposes on my "generic board" (or custom designed board, as was the case).  However, for solving it, I turned the problem inside out from your proposal and used: `-DBOARD_USB_DISC_DEV=NULL -DBOARD_USB_DISC_BIT=NULL` for my compiler options.  And I edited the `board.h` files (on my fork of this repo) to add the `#ifndef BOARD_USB_DISC_DEV` tags around them to avoid "already defined" errors from the compiler in the `board.h` files.  That way, it could default to correct GPIO pins for the board variants using the `USB_DISC` pin and yet can be defined away for projects wanting to disable it, like your  bigtreetech/BIGTREETECH-SKR-mini-E3#144 issue. 

Also, for what it's worth, there is another "USB Discovery" mode in this core that switches the USB pins to GPIO mode and briefly drives the pin low to force the host to rerun the USB discovery.  That code, however, is somewhat strangely `#ifdef` to `GENERIC_BOOTLOADER`  (do a grep on `GENERIC_BOOTLOADER` and you'll find the code I'm referring to -- it's in both the `wirish/boards.cpp` files and in the `usb_setup.cpp` and `usb_generic.c` files of the `USBComposite` library).  I would have thought that that code would have checked the `BOARD_USB_DISC_DEV` and/or `BOARD_USB_DISC_BIT` defines instead as it would need to do that special discovery trigger logic only if the board doesn't have a `USB_DISC` signal and doesn't necessarily have anything to do with whether or not the board has a "generic bootloader" or not.  So I think that code needs to be fixed as well to switch which define it's looking at.

rogerclarkmelbourne, 2020-05-18:
The "Generic boards" don't have the USB discovery hardware. The definition is probably redundant on those variants, except perhaps to save the need for an ifdef in the code to check if its defined or not

closed 2020-05-18 by rogerclarkmelbourne