rogerclarkmelbourne/Arduino_STM32 #761: adding 'libmaple' in platform.txt subject headings

ag88, 2020-03-09:
in stm32duino.com forum, there are often users who installed both official cores and libmaple core.
and they are often confused about which core is it while reporting issues.

i'd like to suggest we add the word libmaple, in the _platform.txt_ name so that they appear in the boards subject headings 
https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/master/STM32F1/platform.txt#L6
https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/master/STM32F4/platform.txt#L6

stevstrong, 2020-10-12:
Roger, what is your opinion on this?

rogerclarkmelbourne, 2020-10-12:
Umm.

Since I have the stm32duino.com domain to STM (free of charge), and STM have been using that name as a trademark even prior to me giving them the domain, I agree it should be changed.

However, I'm not 100% sure what the the name could be 

Possibly just Arduino_STM32  to match this repo ?

I'm not sure everyone will know what "libmaple" means, because its a long time since LeafLabs abandoned this core.

stevstrong, 2020-10-13:
Maybe (Arduino_STM32 aka Libmaple) ?

rogerclarkmelbourne, 2020-10-13:
Perhaps 

Libmaple/ArduinoSTM32

stevstrong, 2020-10-13:
That is promising :)

rogerclarkmelbourne, 2020-10-13:
@ag88 

Comments?

zoomx, 2020-10-14:
LibmapleRCM/ArduinoSTM32
just to be sure not be confused with the original Libmaple?

ag88, 2020-10-14:
Hi Roger, all,Thanks for your response.You can decide on a name you feel comfortable with. My main motivation is simply due to that new users who installed both the official core and this sees stm32duino.com on the board menus in the Arduino IDE. Hence, they confuse this with the official STM core when they post their questions in the forum. As in that when we asked the new user what core are they using, they often reply to that they don't know and yet they are asking questions about the STM32duino cores.Hence, it suffices if we can distinguish this core in the board menus, by asking the user what do they see on the menu. E.g. if they see Arduino_STM32 it is, this core, otherwise it is not. I've originally suggested libmaple as this core is based around libmaple from leaflabs.

stevstrong, 2020-10-14:
I would stick to Roger's suggestion, I think (Libmaple/ArduinoSTM32) would describe correctly the core.

fpistm, 2020-10-14:
From my point of view, stm32duino means stm32 support in Arduino.
stm32duino Libmaple or something like that.
Why not also mention Roger as it relies on your GH username?

rogerclarkmelbourne, 2020-10-14:
It doesn't need to have any mention of me in the name.

I think RCM etc, would just be confusing.

I just did a google search for "libmaple" and was very surprised to find that the original Leaflabs repo , is still top of the list, as are the next 3 hits.

So, this seems to indicate that not many people realise this repo is the active derivative of LibMaple

And perhaps even including LibMaple in the name would be confusing.

As the big difference between this repo and the STM32 official core, is that it doesn't use the HAL, perhaps this could be included in the name. But then again, perhaps hardly anyone would understand "non-HAL"or some similar text

BTW. I never expected people to still be using this core several years after the HAL core was available.

stevstrong, 2020-10-15:
Yes, it seems Libmaple repository to be the first result, but it is arked as retired.
So I think mentioning Libmaple here would not be wrong.
Maybe then (Libmaple-based)?
Or simply (Arduino_STM32)?

fpistm, 2020-10-15:
Don't know if you used the latest Arduino IDE, this is how the menu is displayed now:
[image]

Note that like tools directories have a boards.txt they are displayed.

ag88, 2020-10-17:
In a sense we have maintained and enhanced libmaple core introducing various fixes and improvements. To the credit of the origins, I'd think it is appropriate to mention this as a libmaple core.
I kind of remember one of the fixes for uart I added is to drop bytes with corrupted stop bits or parity errors. This makes various sketches that expects a clean connection work more reliably. Many others have contributed other fixes.

stevstrong, 2020-12-13:
If anyone has problems with that, just post here.

commit 48c16ff3 (2020-12-13): F1/F4: Change board identification text for this core - fixes #761
closed 2020-12-13 by stevstrong