rogerclarkmelbourne/Arduino_STM32 #765: STM32F103C8T6 Serial.print does not work!

denizetkar, 2020-03-17:
**Describe the problem:**  
I cannot get the Serial to print anything.

**Hardware used & connections:**  
STM32F103C8T6, USB-to-UART converters with Prolific PL2303 chip.
STM32F103C8T6 GND <--> converter GND
STM32F103C8T6 3.3v <--> converter 3.3v
STM32F103C8T6 A9 <--> converter RX
STM32F103C8T6 A10 <--> converter TX

**Versions of related software used:**
OS: Windows 10 (version 1909, build 18363.657)
Arduino IDE 1.8.12
Arduino SAM Boards (32-bit ARM Cortex-M3) 1.6.12
[rogerclarkmelbourne/Arduino_STM32](https://github.com/rogerclarkmelbourne/Arduino_STM32/tree/4d355e23763187329f3660f91e78fd40b7df11bc) latest master branch (as of writing this)
PL2303 driver (perfectly compatible with my particular converter)

**Steps to reproduce the problem:**

1. Install Arduino IDE 1.8.12,
2. From boards manager, install Arduino SAM Boards (32-bit ARM Cortex-M3) 1.6.12,
3. Download master branch of this repo, rename it to "Arduino_STM32" and move it to %UserProfile%\Documents\Arduino\hardware,
4. Restart Arduino,
5. Make the pin connections between STM32 and the converter as stated before,
6. Connect the converter to a USB port,
7. Try to upload the following sketch:
```
void setup() {
  Serial.begin(9600);
  Serial.println("Hello world!");
}

void loop() {
}
```

**What is expected?**
When I open the serial monitor of the corresponding COM line, I would expect to see "Hello world!" being printed.

**What happens instead?**
No compilation or upload error happens. The program most likely starts running but there is not a single character printed on the serial monitor.

dewhisna, 2020-03-17:
If you are printing in the `setup()` function, it's probably printed before you even get a chance to launch your serial monitor application.  If the port isn't open, your OS isn't going to capture/buffer the data and it will be lost and you'll never see anything.  And since there's no code in your `loop()`, the board won't do anything else.

Try moving the print to the loop() and insert a delay.  The delay will do two things 1) it will give you time to get your serial monitor tool open, and 2) it will keep from bombarding your serial port with more data than you can keep up with.

In other words, try this:
```C++
void setup() {
  Serial.begin(9600);
}

void loop() {
  delay(10000);
  Serial.println("Hello world!");
}
```

This will print `Hello world!` to the serial port every 10 seconds.

closed 2020-03-17 by denizetkar

rogerclarkmelbourne, 2020-03-17:
Thanks Donna