rogerclarkmelbourne/Arduino_STM32 #767: "ADC' does not name a type" Error on Stm32f103c8 Example of STM32ADC

nikjayswal, 2020-04-09:
/*
  This example shows how to use the ADC library to continuously sample
  one channel/pin.

*/
#include <STM32ADC.h>

ADC myAdc(ADC1);

void setup() {
  Serial.begin(19200);

}

void loop(){

}; //end loop

stevstrong, 2020-04-09:
Obviously, this example does not make too much sense.
But you are welcome to implement it.

dewhisna, 2020-04-09:
I think he's trying to point out that:
https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/4d355e23763187329f3660f91e78fd40b7df11bc/STM32F1/libraries/STM32ADC/examples/SingleChannelContinuousConversion/SingleChannelContinuousConversion.ino#L8

is wrong and should be:

```diff
-ADC myAdc(ADC1);
+STM32ADC myAdc(ADC1);
```

to match the name of the `STM32ADC` class defined in the `STM32ADC.h` header file.

Ledrunning, 2020-04-15:
Single channel ADC with interrupt for Blue pill (taken from original examples, just ported for blue pill pins)
```cpp
#include <STM32ADC.h>
#define TRIGGER PB1
#define BOARD_LED PC13 

STM32ADC myADC(ADC1); 

uint8 pin = PB0; 

volatile static bool triggered = 0; 
uint32 dado_adc = 0; 

void int_func() {
  
  triggered = 1; 
  dado_adc = myADC.getData();
  }

void setup() {
  Serial.begin(19200); 
  pinMode(BOARD_LED, OUTPUT);
  pinMode(TRIGGER, INPUT);
  pinMode(PB0, INPUT_ANALOG);//AD pin.
//startup blink... good idea from Pig-O-Scope
  digitalWrite(BOARD_LED, HIGH);
  delay(1000);
  digitalWrite(BOARD_LED, LOW);
  delay(1000);
  myADC.calibrate();
  myADC.setSampleRate(ADC_SMPR_1_5);
  myADC.attachInterrupt(int_func, ADC_EOC);
  myADC.setPins(&pin, 1);  
  
}

void loop() {
      if(digitalRead(TRIGGER) == 1 ) {
      Serial.println("begin");      
      // Take our samples
      myADC.startConversion(); 
      while (triggered == 0); //wait here... 
      
      Serial.print("Readin: ");
      Serial.println(dado_adc);      
      }

}
```

stevstrong, 2020-12-07:
Fixed with https://github.com/rogerclarkmelbourne/Arduino_STM32/commit/b6eb20d00a4010d83b386b2cff830a363cfebc34.

closed 2020-12-07 by stevstrong