rogerclarkmelbourne/Arduino_STM32 #768: I can't write and read data on my stm32f103c8 microcontroller.

Joypacharya, 2020-04-11:
Hello !!!!
Actually I want to write and read data from EEPROM (Flash Emulation).
But I could not get data while I saved it into specific address.
I hope you will help me to resolve my problem.
Thank you,

Joypacharya, 2020-04-11:
Here I am using for code 
#include <EEPROM.h>

int i,j;
int ovfi,ovfj;
int value1 = 0,value2 = 0;
void setup() 
{
  Serial2.begin(115200);      //USART2_TX_PIN => PA2 , USART2_RX_PIN => PA3
  Serial2.println("Serial2 init successful! V2.0");
  pinMode(PC13, OUTPUT);
  value1 = EEPROM.read(0, (uint16*)&ovfi);
  value2 = EEPROM.read(10, (uint16*)&ovfj);
  Serial2.print("Value 1 :");
  Serial2.print(value1);
  Serial2.print(" Value 2 :");
  Serial2.println(value2);
  delay(1000);
}

void loop() 
{
  Serial2.print("Value 1 :");
  Serial2.print(value1);
  Serial2.print(" Value 2 :");
  Serial2.println(value2);
  ovfi = i+1;
  ovfj = j+1;
  EEPROM.write(0,ovfi);
  EEPROM.write(10,ovfj);
  delay(1000);
  value1 = EEPROM.read(0, (uint16*)&ovfi);
  value2 = EEPROM.read(10, (uint16*)&ovfj);
  Serial2.print("Value 1 :");
  Serial2.print(value1);
  Serial2.print(" Value 2 :");
  Serial2.println(value2);
  delay(1000);
}

stevstrong, 2020-10-12:
Please use the example for that library.
Also, please use the forum for such inquiries: www.stm32duino.com

closed 2020-10-12 by stevstrong