rogerclarkmelbourne/Arduino_STM32 #769: ADC in DMA mode can not reading sample

Ledrunning, 2020-04-15:
Hi there! I try your example MultiChannelSingleConversion.ino and got a problem.
I use Blue Pill with STM32F103C8. So I've changed yor defines for this board then comiled and uploaded scetch.
When I triggered my digital pin I saw only "begin" in on my serial monitor...
I think program keep crushing on while loop and does't work beyond because i didn't see string "End of scanning"
I have the same problem with [PigOScope](https://github.com/pingumacpenguin/STM32-O-Scope/issues/13) . Why both programs does't exit the while loop with volatile flag to change routine state?

stevstrong, 2020-04-15:
I see that the example uses pin 11 for sampling. This is an USB pin, so it may cause a conflict on USB, that can be a reason why you cannot see any data coming form the board.
Try to use less pins, let say only PA0 till PA3. Configure the array `pins[]` accordingly.

Ledrunning, 2020-04-15:
Thanks for your reply! So I changed my code for one input channel if am I right:
```cpp
#include <STM32ADC.h>

#define BOARD_LED PC13 //PB0
#define TRIGGER PB1

uint8 analogInPin = PB0;
const int maxSamples = 1; // 8 channels 

// Array for the ADC data
uint16 dataPoints[maxSamples]; 

volatile static bool dma1_ch1_Active; //flag for interrupt

STM32ADC myADC(ADC1);

void setup() {
//start Serial
  Serial.begin(19200);
//Start up blink from Pig-O-Scope
  pinMode(BOARD_LED, OUTPUT);
  pinMode(TRIGGER, INPUT);
  digitalWrite(BOARD_LED, HIGH);
  delay(1000);
  digitalWrite(BOARD_LED, LOW);
  delay(1000);
//calibrate ADC.
  myADC.calibrate(); 
  
  myADC.setSampleRate(ADC_SMPR_1_5); //sample ratio
  myADC.setPins(&analogInPin, maxSamples);   //pins to be converted
  myADC.setScanMode();               //Set the ADC in Scan Mode
}

void loop(){
//start acquisition on button push. 
    if(digitalRead(TRIGGER) == 1 ) {
        Serial.println("begin");
        dma1_ch1_Active = 1;
        myADC.setDMA(dataPoints, 1, (DMA_MINC_MODE | DMA_TRNS_CMPLT), DMA1_CH1_Event); 
        myADC.startConversion();
        //while (dma1_ch1_Active == 1); //wait for DMA to complete.  
        Serial.println("End of scanning");
        
        for(unsigned int i = 0; i < maxSamples; i ++) {
            Serial.print("sample[");
            Serial.print(i);
            Serial.print("] = ");
            Serial.println(dataPoints[i]);
        }
      while(digitalRead(D32) == 1); //stay here. Another button push is required. 
      }
}; //end loop

/*
    Interrupt handler. eh eh 
*/
static void DMA1_CH1_Event() {
  dma1_ch1_Active = 0;
}
```
Have the same problem. What I'm doing wrong?

Ledrunning, 2020-04-21:
I tried different variation, but the example with DMA didn't work for me. With last release and current Master branch in Arduino IDE. I compiled in 1.8.1 version and last Arduino IDE released.

stevstrong, 2020-06-11:
I think the example is broken, I attach one which should work (I have tested).
For that, you also need to place the attached two files (from ZIP) in the STM32ADC library src directory.
Let me know if it works now.
[MultiChannelSingleConversion.ino.txt](https://github.com/rogerclarkmelbourne/Arduino_STM32/files/4765499/MultiChannelSingleConversion.ino.txt)
[src.zip](https://github.com/rogerclarkmelbourne/Arduino_STM32/files/4765507/src.zip)

Ledrunning, 2020-06-16:
Thank you for your reply! I used your example but unfortunately the example did not work. The LED lights up continuously, on the serial monitor screen there is only the inscription: "MultiChannelSingleConversion demo started begin". I use the latest Stm32 kernel release, arduino IDE 1.8.1 and the BluePil board. I feed 3.3V to pin from A0 to A7 and there is no data in the terminal.

stevstrong, 2020-06-17:
Did you replace also the source files attached?
I assume so, otherwise you should have get some errors at compilation.

Here is my output:
```
MultiChannelSingleConversion demo started

begin
sample[0] = 1578
sample[1] = 1901
sample[2] = 2640
sample[3] = 2664
sample[4] = 2684
sample[5] = 2694
sample[6] = 2151
sample[7] = 2112
```

Ledrunning, 2020-06-17:
What version of core do you use? I try latest with your example and nothing to changed doesn't even LED blink(PC13) then 1.0.0 with compilation warning `warning: undefined reference to `adc_dma_enable(adc_dev*)'` but it does not work correctly just blinking. Examples based on interrupt works fine on latest core in main branch.
My output just:
MultiChannelSingleConversion demo started:
begin
Thats all. I connected PA0, PA1, PA2 etc to 3.3V Maybe my Blue Pill is broken...

stevstrong, 2020-06-18:
I will post my working binary file later today.

Ledrunning, 2020-06-18:
Okay! Thank you! I'm waiting...

davidkleng, 2020-10-02:
@Ledrunning Did you get it to work? If you, like me, have the stm32 "Blue pill", this example should work. 
I use the "STM32duino". 

`/*
  This example shows how to use the ADC library to sample several channels/pins in one go. 
  This example attaches an interrupt to the DMA completion to notify the user. 

*/
#include <STM32ADC.h>

#define BOARD_LED PC13

uint8 pins[] = {PA0,PA1,PA2,PA3,PA4,PA5,PA6,PA7};

const int maxSamples = 8; // 8 channels 

// Array for the ADC data
uint16 dataPoints[maxSamples]; 

volatile static bool dma1_ch1_Active; //flag for interrupt

STM32ADC myADC(ADC1);

//-----------------------------------------------------------------------------
//  Interrupt handler.
//-----------------------------------------------------------------------------
static void DMA1_CH1_Event()
{
	dma1_ch1_Active = 0;
}
//-----------------------------------------------------------------------------
void setup()
{
	//start Serial
	Serial.begin(19200);
	pinMode(BOARD_LED, OUTPUT);
	digitalWrite(BOARD_LED, HIGH);

	while(!Serial); delay(10);

	Serial.println("MultiChannelSingleConversion demo started\n");
	// calibrate ADC.
	myADC.calibrate(); 
	// setup ADC
	myADC.setSampleRate(ADC_SMPR_1_5); //sample ratio
	myADC.setPins(pins, maxSamples);   //pins to be converted
	myADC.setScanMode();               //Set the ADC in Scan Mode
	myADC.setDMA(dataPoints, (DMA_MINC_MODE | DMA_TRNS_CMPLT), DMA1_CH1_Event);
}
//-----------------------------------------------------------------------------
void loop()
{
// start acquisition
	Serial.println("begin");
	dma1_ch1_Active = 1;

	myADC.startDMA(maxSamples);
	myADC.startConversion();

	while (dma1_ch1_Active == 1); //wait for DMA to complete.  
	// print out the sampled values
	for(unsigned int i = 0; i < maxSamples; i ++) {
		Serial.print("sample[");
		Serial.print(i);
		Serial.print("] = ");
		Serial.println(dataPoints[i]);
	}
}`

stevstrong, 2020-12-13:
I attach here a working binary (upload with STLink).
[MultiChannelSingleConversion.ino.bin.zip](https://github.com/rogerclarkmelbourne/Arduino_STM32/files/5685193/MultiChannelSingleConversion.ino.bin.zip)

I will close this because it works as expected.

closed 2020-12-13 by stevstrong