rogerclarkmelbourne/Arduino_STM32 #776: Core is missing pinMode() mode macros

bperrybap, 2020-05-13:
Arduino cores use #define macros for the pinMode(pin,mode) mode parameter.
These are actually important as sketch and library code can use them to compile time check for the existence of mode capabilities.
i.e. the code could use an ifdef to see if INPUT_PULLUP exists to determine if that capability exists.

This core is using an WiringPinMode enum in io.h instead of using macros.
This breaks code that depends on using compile time detection of pinMode capabilities.

There is an easy fix for this.
Simply create a #define macro for each enum.
This is a great solution as it not only retains the enum, but also provides full backward compatibility as well as forward compatibility.
 i.e once this has been done, code will be able to compile time detect the existence of all the modes including the expanded mode capabilities in this core such as OUTPUT_OPENDRAIN, INPUT_PULLDOWN, INPUT_FLOATING, etc....

All that is necessary is to create a #define macro for each enum like this:
```
#define OUTPUT1 OUTPUT1
#define OUTPUT_OPEN_DRAIN1 OUTPUT_OPEN_DRAIN1
#define INPUT1 INPUT1
#define INPUT_ANALOG1 INPUT_ANALOG1
#define INPUT_PULLUP1 INPUT_PULLUP1
#define INPUT_PULLDOWN1 INPUT_PULLDOWN1
#define INPUT_FLOATING1 INPUT_FLOATING1
#define PWM1 PWM1
#define PWM_OPEN_DRAIN1 PWM_OPEN_DRAIN1
```

rogerclarkmelbourne, 2020-05-14:
This looks problematic to me, because it looks like a circular re-definition

I'm a bit surprised the pre-processor accepts this.

bperrybap, 2020-05-14:
It is not a circular / recursive re-definition. That isn't how macro scanning and expansion works.
This type of definition has always worked and is explicitly allowed by the C standard.
https://gcc.gnu.org/onlinedocs/cppinternals/Macro-Expansion.html
https://gcc.gnu.org/onlinedocs/cpp/Self-Referential-Macros.html#Self-Referential-Macros

From the self referential macro page:

> One common, useful use of self-reference is to create a macro which expands to itself. If you write
> 
> #define EPERM EPERM
> 
> then the macro EPERM expands to EPERM. Effectively, it is left alone by the preprocessor whenever it’s used in running text. You can tell that it’s a macro with ‘#ifdef’. You might do this if you want to define numeric constants with an enum, but have ‘#ifdef’ be true for each constant.

stevstrong, 2020-05-15:
Can you please post a pull request?

bperrybap, 2020-05-15:
It is a trivial update, but ok.
So which branch do you want this based off of?

closed 2020-05-18 by rogerclarkmelbourne

bperrybap, 2020-05-18:
Wow,
You closed the issue with no comment after I was offering to do a PR to resolve this and was just asking which branch you preferred as the base for the PR?
This repository has a TON of branches, I figured it might be helpful to check to see if there was a specific one that you preferred rather than just assume master.
Why would you not want these types of incompatibilities fixed?

stevstrong, 2020-05-18:
Honestly, I do not see any incompatibility, except maybe with your application.

bperrybap, 2020-05-18:
The incompatibility is for code that runs on **MANY** different cores that compile time detects capabilities of a core so that it can automatically adjust/adapt to those capabilities.
i.e. if certain capabilities don't exist, the code can use alternate methods, or worst case spit out a specific error indicating that it requires a feature that the core does not implement.

This is much better than having code that just fails to compile when a symbol doesn't exist or code that has to reduce itself to the lowest common denominator of capabilities across the cores.

In this specific case the symbols and the capability do exist in the core but code that uses the core has no way to detect it at compile time.

When a core does not create the defines, any code that does compile time checks will fail to detect the capability.

Another well known example of using compile detection for capabilities is checking for a built in LED.
This can done by using the variant define LED_BUILTIN.
Since a built in LED is optional, code that may want to use an on board LED must have a way detect its presence and not depend on it.
A core that uses an alternate symbol name or that creates the symbol as a const int without also creating a LED_BUILTIN define, will not allow the code to compile time detect the existence of a built in LED.