rogerclarkmelbourne/Arduino_STM32 #777: SDA and SCL don't exist for Wire library

bperrybap, 2020-05-14:
The defines SDA and SCL in Arduino cores represent the pins used by the Wire library.
This core defines them in SoftWire.h which is only used when using SoftWire.
This causes code that depends on them when using Wire to break.
Should you ask "Why would code using Wire need to know the pins?"
The reason, at least in my case, is I have an i2c diagnostic tool that runs checks on the signals to see if there are shorts, signals stuck to ground, and checks to see if pullups are on the signals.
Not having the SDA and SCL symbols means that the code cannot even compile.

Given that SDA and SCL are normally the hardware pins and they also represent the hardware pins in this core, it seems that a simple solution would be to move the SDA and SCL defines out of SoftWire.h and move them to utility/WireBase.h
That way they exist/work like they do in all the other Arduino cores and will be available to both Wire and SoftWire

stevstrong, 2020-05-14:
The simplest solution is to insert in your code
```
#ifndef SCL
#define SCL PXy // put here the right pin
#endif
```
I don't see why this would be beneficial to the core at all, except your diagnostic tool.

bperrybap, 2020-05-14:
@stevstrong 
What you have shown isn't a workable solution for s/w that needs to run and "just work" on any/all Arduino platforms without any user modifications.
The code currently runs on any Arduino platform that has a Wire library. 
This is hundreds of boards and a dozen+ cores. 
Except the code won't compile on this core.....

To flip things, I don't see how not properly defining SDA and SCL is beneficial to this core.
Every other core that has a Wire library and I mean EVERY other core defines SDA and SCL, including the other STM32 core from this board package URL: 
https://github.com/stm32duino/BoardManagerFiles/raw/master/STM32/package_stm_index.json
What benefit is there to being different from every other Arduino core?

Other cores define SDA and SCL in the variant file.
This allows setting up the proper pin definitions for different boards/processors that may be using the same core.
Ideally, this core should also create the SDA and SCL defines in the variant file.
However, since it appears that it is using the same pins for all the boards in the same core it could choose to put the definitions somewhere else.
Technically, the symbols should **always** exist. Not just when Wire.h or some other i2c header is included so my previous suggestion of moving the defines to utility/WireBase.h is more of a hack than a true solution.

The problem with trying to use conditionals up in a sketch or library to define these symbols is that it rapidly explodes into a never ending maintenance issue.
The code would to need to know which core, which variant,  and which board the code is being compiled for to able to pick the proper pin.
This is gets very complicated when dealing with cores/variants/boards as each variant and each board can potentially define or re-define its own values.
Even if the code did attempt to try to do this, compile time detecting which core and then which variant and then which board is being used is extremely complicated and would require lots of conditionals and in some cases it is not possible to detect between different boards within a given core or variant.
I am very familiar with trying to do this as I have tried to do it on another project, and it is so so at best as it isn't always possible to detect things all the way down to which individual board has been selected, especially since users can modify or create additional boards in their boards.txt file.
Then to maintain it is a never ending nightmare.

stevstrong, 2020-05-15:
Can yo please indicate where the Arduino framework prescribes the existence of those defines?
This core existed several years so far without those defines.
Each core is different and can chose which terms to define or not, as long as Arduino framework compatible.
Anyway, I already can imagine problems when users try to remap the I2C pins and that does not match the define.
However, if you would like to contribute, please submit a pull request.

bperrybap, 2020-05-15:
And many Arduino variant/board defines/symbols that are missing in this platform/core have existed for many years before this platform/core existed.

> Anyway, I already can imagine problems when users try to remap the I2C pins and that does not match the define.

This makes no sense as there are already defines for SDA and SCL in the core, it is just that they are only created when SoftWire.h is included.
The SDA and SCL symbols should exist all the time like they do in all the other cores.
They represent the digital pin numbers of the pins used by the Wire library.
These symbols go back to the Arduino 1.0 release from 2011.

You know there is no such formal documentation. Arduino has never been that well documented.
There are literally hundreds of symbols and macros in the Arduino system.
There are variant/board specific symbols like NUM_DIGITAL_PINS, NUM_ANALOG_PINS, the An symbols, the newer (from back in 2016) PIN_xxx symbols,  the LED_BUILTIN define, and on and on
And some of them must be defines not const int (this was discussed at length on the developers email list several years back)

Many new board specific symbols were created/added when the IDE platform tree was modified to support variants which was part of the 1.0 release back in Nov 2011 
here is a comment from the release 1.0 notes:

>   The new, variant-specific pins_arduino.h files now provides additional
>   macros with information about the characteristics of the board (e.g.
>   the locations of the SPI and TWI pins, and the number of digital and
>   analog pins).

I believe that this may be the closest thing to actual documentation that exists.

SDA and SCL were also added at this point in time.

Given the lack of formal documentation, the best you can do is look at the Arduino.cc files like Arduino.h and the variant files and try to replicate what they have done.
The missing SDA and SCL defines is just one of several issues that users are starting to experience with this platform/core do to Arduino incompatibilities.

The best thing to do would be to look at the Arduino "standard" variant and make sure you have implemented all the symbols created in that file.
https://github.com/arduino/ArduinoCore-avr/blob/master/variants/standard/pins_arduino.h
This is what the other 3rd party platform/core writers have done.

The cores in this platform are definitely the odd man out as all the other cores including the STM32 platform have implemented  the SDA and SCL symbols as well as other symbols that are also missing in this platform and its variants.

rogerclarkmelbourne, 2020-05-18:
@bperrybap  please use another core if this one does not fit your use requirement, or submit a PR and show how how your PR will not affect any of the thousands of other people who use this core

closed 2020-05-18 by rogerclarkmelbourne

bperrybap, 2020-05-18:
It isn't me that is having issues with the core.
It is others. Other people are having issues with the core's incompatibilities.
This includes issues that are unique to this core like missing symbols such as SDA and SCL.
I got pulled in because someone wants to use one of my Arduino libraries and a diagnostic tool on this core.
It runs on every other Arduino core but fails to compile on this core due to missing variant symbols.

If you like, I'll generate a PR that at least moves the SDA and SCL out of SoftWire. h and into utility/WireBase.h where it should be.
The reasoning being that since SDA and SCL represent the hardware pins used by Wire (at least that is what the comment says), those defines should at least be moved into WireBase.h so that they can exist for both Wire and SoftWire libraries.

stevstrong, 2020-05-18:
There is basic difference between SofWire and HardWire.
You can define any GPIO as SDA and SCL in SofWire, while that does not make any sense  for HardWire.
I am curious how you would solve this incompatibility. also taking into account the possibility to remap the wire pins.

bperrybap, 2020-05-18:
Perhaps there is some miscommunication about what I mean by **SCL** and **SDA**.
I just want to make sure we are talking about the software symbols **SDA** and **SCL** not actual i2c signal pins.
On every other core, **SDA** and **SCL** are the names of symbols that sketches and libraries can use/reference that represent the Arduino digital pin numbers used by the Wire library for the i2c sda and scl signals.
On boards that implement i2c in hardware, the symbols represent the Arduino digital pin numbers for the hardware i2c signals on the variant/board.

In cores that allow changing the pins used by the Wire library, like the ESP8266 and ESP32 cores, SDA and SCL represent the Arduino pin numbers used by default and are not changed should the user alter the pins being used from their default values.