rogerclarkmelbourne/Arduino_STM32 #778: F_CPU is not defined in sketches

bperrybap, 2020-05-14:
For some reason the macro F_CPU does not seemed to be defined in sketches.
I see it being used in some of the library code so perhaps it is defined but not is being handed to sketches.
Typically it is defined on the command line handed to the compiler.

stevstrong, 2020-05-14:
> I see it being used in some of the library code so perhaps it is defined but not is being handed to sketches.

How do you check this?

> Typically it is defined on the command line handed to the compiler.

It is defined in the boards.txt, is used to compile process, and it works for sketches as well.

bperrybap, 2020-05-14:
I wasn't checking it. The sketch code was simply printing it.
The symbol wasn't being defined for sketches for the board STM32 board I picked.
I went back an looked closer. Turns out it shows up for the STM32F1 boards, but not the STM32F3 or STM32F4 boards.
The neither the boards.txt nor platform.txt files in those packages have support for F_CPU so it isn't being defined in those core packages.

Here is simple test sketch that will demonstrate the issue:
Try it on the "blackpill" board or the "STM32F3Discovery" board
```
void setup(void)
{
	Serial.begin(9600);
}
void loop(void)
{
	Serial.print(F_CPU);
	delay(1000);
}
```

It outputs this compile error:
```
/home/bill/Documents/devel/Arduino/sketchbook/sketches/test/tmp/stm32fcpu/stm32fcpu.ino: In function 'void loop()':
stm32fcpu:8:15: error: 'F_CPU' was not declared in this scope
  Serial.print(F_CPU);
               ^
exit status 1
'F_CPU' was not declared in this scope
```

commit faa0dc5e (2020-05-15): F4: fix #778
closed 2020-05-15 by stevstrong