rogerclarkmelbourne/Arduino_STM32 #779: CMD only works in Windows platform

apung, 2020-05-14:
Hello awesome people,

I think we have a little problem here:
https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/f6fb21ace2dadbf8eedb418409ed5346ffdca6aa/STM32F4/platform.txt#L91

For a while, I just remove (add comment character) that line.

stevstrong, 2020-05-14:
I cannot see any problem on Windows.

bperrybap, 2020-05-14:
Uh yea. This is totally broken.
@stevstrong 
Sure it works on windows, but the point is that platform.txt cannot use recipes that are hardcoded to only work on Windows.

bperrybap, 2020-05-14:
BTW, this issue does not exist for the boards in the STM32F1 core package.
There are MANY differences between the two platform.txt files including in some in areas where it seems like they should be the same.

closed 2020-05-15 by stevstrong

apung, 2020-05-15:
The point of this issue is:
https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/f6fb21ace2dadbf8eedb418409ed5346ffdca6aa/STM32F4/platform.txt#L91 only works in Windows.

Sure, you have no problem at all while using Windows. But for another platform like Linux or MacOS, the command will get a huge problem. Why? Because there's **no cmd** in Linux or MacOS. @stevstrong 

I resolve this problem by simply add comment that line.
```
## Create assembly S file
# recipe.objcopy.lst.pattern=cmd /C "{compiler.path}{compiler.elf2lst.cmd}" {compiler.elf2lst.flags} "{build.path}/{build.project_name}.elf" > "{build.path}/{build.project_name}.lst"
```

@bperrybap in STM32F1, there's no
```
## Create assembly S file
```
so you can build the platform with no problem.

Btw, thanks for the great project 👍 💯 🥇

commit d5521f0e (2020-05-15): F4: fix #779