rogerclarkmelbourne/Arduino_STM32 #780: The code doesn't compile for STM32F401CCU6 Black Pill Board

jhsa, 2020-05-15:
The code doesn't compile for STM32F401CCU6 Black Pill Board, even the i2c scanner example does not compile. Please fix if you can. Here is the error shown by the arduino IDE. I have the latest core installed.

C:\Users\Joao\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F4\2020.4.23\libraries\Wire\Wire.cpp: In constructor 'TwoWire::TwoWire(uint8, uint8)':

C:\Users\Joao\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F4\2020.4.23\libraries\Wire\Wire.cpp:66:20: error: 'I2C2' was not declared in this scope

         sel_hard = I2C2;

stevstrong, 2020-05-15:
Solved with 1b54573.

closed 2020-05-15 by stevstrong

jhsa, 2020-05-15:
Thanks, will try a bit later.. Is there a new version out, or do I have to install manually??

stevstrong, 2020-05-15:
You can take the [new Wire.cpp](https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/master/STM32F4/libraries/Wire/Wire.cpp) and copy to your local file directory.

jhsa, 2020-05-15:
Thank you, done.