rogerclarkmelbourne/Arduino_STM32 #781: libmaple/usb.h not found

jhsa, 2020-05-15:
Hi, I thought this core supported libmaple like the STM32F1 does.. Obviously it is not the case? Or am I doing something wrong? If not supported, any change it would be in the future? Below is the related IDE output error. Thanks

```

In file included from C:\Users\Joao\Documents\Arduino\libraries\USBComposite_stm32f1/USBComposite.h:6:0,

                 from E:\Electronic Projects\Audio\Loop Pedals\iLoopino\New Version using a Byte per Preset\iLoopino STM32F401_Based on Version 87_Direct_To\iLoopinoSTM32F401_New_Saving\iLoopinoSTM32F401_140_DIY_LCD_BackPack\iLoopinoSTM32F401_140_DIY_LCD_BackPack.ino:205:

C:\Users\Joao\Documents\Arduino\libraries\USBComposite_stm32f1/usb_generic.h:7:26: fatal error: libmaple/usb.h: No such file or directory

 #include <libmaple/usb.h>

                          ^

compilation terminated.
```

stevstrong, 2020-05-15:
You are not using Arduino IDE, right?
Yo have to make sure that the path 
`"...\\Arduino\\hardware\\Arduino_STM32\\STM32F1\\system/libmaple"`
is included during the compilation.

jhsa, 2020-05-15:
I am using the IDE.. Perhaps my post above wasn't clear enough, I apolozise. I am compiling for the STM32F401CCU6 board.

stevstrong, 2020-05-15:
Please post the verbose build output to see which options are used during compilation.

jhsa, 2020-05-15:
This??

`E:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware E:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Joao\AppData\Local\Arduino15\packages -hardware C:\Users\Joao\Documents\Arduino\hardware -tools E:\Program Files (x86)\Arduino\tools-builder -tools E:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Joao\AppData\Local\Arduino15\packages -built-in-libraries E:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Joao\Documents\Arduino\libraries -fqbn=stm32duino:STM32F4:blackpill_f401:usb_cfg=usb_serial,opt=osstd -ide-version=10812 -build-path C:\Users\Joao\AppData\Local\Temp\arduino_build_450951 -warnings=none -build-cache C:\Users\Joao\AppData\Local\Temp\arduino_cache_7871 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\Joao\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.arm-none-eabi-gcc-4.8.3-2014q1.path=C:\Users\Joao\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.stm32tools.path=C:\Users\Joao\AppData\Local\Arduino15\packages\stm32duino\tools\stm32tools\2020.4.23 -prefs=runtime.tools.stm32tools-2020.4.23.path=C:\Users\Joao\AppData\Local\Arduino15\packages\stm32duino\tools\stm32tools\2020.4.23 -verbose E:\Electronic Projects\Audio\Loop Pedals\iLoopino\New Version using a Byte per Preset\iLoopino STM32F401_Based on Version 87_Direct_To\iLoopinoSTM32F401_New_Saving\iLoopinoSTM32F401_140_DIY_LCD_BackPack\iLoopinoSTM32F401_140_DIY_LCD_BackPack.ino
E:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware E:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Joao\AppData\Local\Arduino15\packages -hardware C:\Users\Joao\Documents\Arduino\hardware -tools E:\Program Files (x86)\Arduino\tools-builder -tools E:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Joao\AppData\Local\Arduino15\packages -built-in-libraries E:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Joao\Documents\Arduino\libraries -fqbn=stm32duino:STM32F4:blackpill_f401:usb_cfg=usb_serial,opt=osstd -ide-version=10812 -build-path C:\Users\Joao\AppData\Local\Temp\arduino_build_450951 -warnings=none -build-cache C:\Users\Joao\AppData\Local\Temp\arduino_cache_7871 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\Joao\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.arm-none-eabi-gcc-4.8.3-2014q1.path=C:\Users\Joao\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.stm32tools.path=C:\Users\Joao\AppData\Local\Arduino15\packages\stm32duino\tools\stm32tools\2020.4.23 -prefs=runtime.tools.stm32tools-2020.4.23.path=C:\Users\Joao\AppData\Local\Arduino15\packages\stm32duino\tools\stm32tools\2020.4.23 -verbose E:\Electronic Projects\Audio\Loop Pedals\iLoopino\New Version using a Byte per Preset\iLoopino STM32F401_Based on Version 87_Direct_To\iLoopinoSTM32F401_New_Saving\iLoopinoSTM32F401_140_DIY_LCD_BackPack\iLoopinoSTM32F401_140_DIY_LCD_BackPack.ino
Using board 'blackpill_f401' from platform in folder: C:\Users\Joao\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F4\2020.4.23
Using core 'maple' from platform in folder: C:\Users\Joao\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F4\2020.4.23
Detecting libraries used...
"C:\\Users\\Joao\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -Wall -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DERROR_LED_PIN=PC13 -w -x c++ -E -CC -mcpu=cortex-m4 -DARDUINO=10812 -DVARIANT_blackpill_f401 -DARDUINO_ARCH_STM32F4 -DLED_BUILTIN=PC13 -DCRYSTAL_FREQ=25 -DNO_CCMRAM -DSERIAL_USB -DUSER_ADDR_ROM=0x08000000 -mthumb -D__STM32F4__ -DSTM32F4 "-IC:\\Users\\Joao\\AppData\\Local\\Arduino15\\packages\\stm32duino\\hardware\\STM32F4\\2020.4.23\\system/libmaple" "-IC:\\Users\\Joao\\AppData\\Local\\Arduino15\\packages\\stm32duino\\hardware\\STM32F4\\2020.4.23\\cores\\maple/libmaple/usbF4" "-IC:\\Users\\Joao\\AppData\\Local\\Arduino15\\packages\\stm32duino\\hardware\\STM32F4\\2020.4.23\\cores\\maple" "-IC:\\Users\\Joao\\AppData\\Local\\Arduino15\\packages\\stm32duino\\hardware\\STM32F4\\2020.4.23\\variants\\blackpill_f401" "C:\\Users\\Joao\\AppData\\Local\\Temp\\arduino_build_450951\\sketch\\iLoopinoSTM32F401_140_DIY_LCD_BackPack.ino.cpp" -o nul
Alternatives for Adafruit_MCP23008.h: [Adafruit_MCP23008_library@1.0.2]
ResolveLibrary(Adafruit_MCP23008.h)
  -> candidates: [Adafruit_MCP23008_library@1.0.2]
"C:\\Users\\Joao\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -Wall -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DERROR_LED_PIN=PC13 -w -x c++ -E -CC -mcpu=cortex-m4 -DARDUINO=10812 -DVARIANT_blackpill_f401 -DARDUINO_ARCH_STM32F4 -DLED_BUILTIN=PC13 -DCRYSTAL_FREQ=25 -DNO_CCMRAM -DSERIAL_USB -DUSER_ADDR_ROM=0x08000000 -mthumb -D__STM32F4__ -DSTM32F4 "-IC:\\Users\\Joao\\AppData\\Local\\Arduino15\\packages\\stm32duino\\hardware\\STM32F4\\2020.4.23\\system/libmaple" "-IC:\\Users\\Joao\\AppData\\Local\\Arduino15\\packages\\stm32duino\\hardware\\STM32F4\\2020.4.23\\cores\\maple/libmaple/usbF4" "-IC:\\Users\\Joao\\AppData\\Local [... cut ...]
                          ^

compilation terminated.

Using library Adafruit_MCP23008_library at version 1.0.2 in folder: C:\Users\Joao\Documents\Arduino\libraries\Adafruit_MCP23008_library 
Using library Wire in folder: C:\Users\Joao\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F4\2020.4.23\libraries\Wire (legacy)
Using library USBComposite_stm32f1 at version 0.98 in folder: C:\Users\Joao\Documents\Arduino\libraries\USBComposite_stm32f1 
exit status 1
Error compiling for board Blackpill STM32F401CCU6.
`

dewhisna, 2020-05-15:
Seems like you are mixing F1 and F4 code bases.  I'm not sure how well that works.  The two families have similar peripherals, but not everything is directly interchangeable, especially the low-level drivers.

stevstrong, 2020-05-15:
F4 core is not compatible with USB composite library for F1.

closed 2020-05-15 by stevstrong

jhsa, 2020-05-15:
ok, thanks.. By any chance, do you know any USB MIDI library that runs on the F401?

stevstrong, 2020-05-15:
No, but g**gle is your friend.

jhsa, 2020-05-15:
Oh, don't I know that? :) Why do you think I have just asked you? Because I can't find anything on g**.
The one I was trying to is the only one I found that did work on the F103. So, no hope having USB MIDI on the F4, at least not with arduino. 
Thanks a lot for your help..

dewhisna, 2020-05-15:
@jhsa  Having worked a little with a F405 on the NetduinoPlus2 board and pulling me fur out trying to get USB working on it (and finally giving up), there seems to be quite a few differences in the USB peripheral between the F1 and F4 (and even some differences across members within the F4 family).  According to [this ST Community Forum Post](https://community.st.com/s/question/0D50X00009XkXHj/stm32f4-as-a-usbmidi-device) a possible lead for USB MIDI on the STM32F4 is [this Dekrispator_v2 repo](https://github.com/MrBlueXav/Dekrispator_v2).  It looks like it uses ST's HAL code from their STM32Cube software.  With a bit of work, you might can pull enough of it into your code to make it work -- it depends on how locked into the Arduino API you are for doing your USB MIDI or if your other code is flexible enough to be tailored for this.

Another possibility if you don't need an actual MIDI Audio Class USB device (i.e. where your board appears as an actual MIDI adapter of some sort), you can use the Virtual Serial port of the USB and stream it that way, since MIDI itself is just a serial stream.  It would still be USB, only as a serial device instead of audio device.  If you did that, you'd need raw serial MIDI processing code instead of USB Audio Class code, as the latter adds the concept of multiple cables and transports the messages in special 32-bit USB frames.

jhsa, 2020-05-15:
Thanks for your help.. I am a beginner, so I couldn't use that ST HAL code :) 
I do need a real USB midi. It could be HID, but better if it could also work as an USB Host.  But I won't be able to do it unless there was already an arduino library for it with some examples.. I got serial midi working on it, and I can do it also the way you described, using the USB serial. But that won't work with the devices I want to connect it to, as they are expecting real USB. 
Thanks a lot for your help anyway. Very much appreciated..