rogerclarkmelbourne/Arduino_STM32 #785: STM32ADC Examples

nikjayswal, 2020-05-23:
Please ADD those 2 codes in STM32ADC DMA Examples.

dma_disable(DMA1, DMA_CH1); //Disable DMA and transfer

myADC.setTrigger(ADC_EXT_EV_SWSTART); // Disable ADC & DMA Trigger

nikjayswal, 2020-05-23:
and Also PC13 LED Working inverted if i write High the LED is Off and if i write LOW led On.

dewhisna, 2020-05-23:
@nikjayswal -- PC13 is working backwards because that's exactly how the blue pill board is designed. The PC13 pin can't source much current.  It can, however, barely sink enough current to turn on an LED at really low power.  So that's what they did in the design.  Therefore, writing it to HIGH is Off, and writing it to LOW is On.  That is NOT a code problem but because the designers of the blue pill were too cheap to add a transistor between the CPU and the LED to provide enough current flow and not run it inverted.  If you don't want it to work backwards, then redesign your board so that the LED isn't wired inverted.

stevstrong, 2020-05-23:
@nikjayswal , in which example do you mean?

nikjayswal, 2020-05-24:
STM32ADC "SingleChannelAtSampleRateCircularBuffer , MultiChannelSingleConversion , MultiChannelContinuousConversion "

Those two lines is useful to disable the DMA and ADC Trigger to avoid analogRead errors "please add those two lines as a comment in Example sketch".

stevstrong, 2020-05-24:
Can you make a pull request?

stevstrong, 2020-10-12:
I will close this due to inactivity.

closed 2020-10-12 by stevstrong