rogerclarkmelbourne/Arduino_STM32 #793: stm_fft slow blink PC13 on bluepill

78edu, 2020-05-31:
Hello.
I've tryed stm_fft library and it doesn't work.
I've installed Arduino_STM32 by an archive to my Documents folder and Arduino SAM boards in Board Manager.
All works fine, but not this library.
Arduino uses: "C:\\Users\\1\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-g++"
And no *.asm file is compiled.
With .s extension it compiles as with -o nul:
"...\\libraries\\stm_fft\\cr4_fft_256_stm32.s" -o nul"
I have no idea how to properly compile this files and link it.
I've tried info from other issues on this topic with no result. STM32duino toolchain is different and i don't know how to use it, it fails in many places.
What should i do to use this library?

78edu, 2020-05-31:
After installing 1.6.1 arduino uses arm-none-eabi-gcc 8.2.1-1.7, but i still have pulsating led issue.
.asm files also ignored, but not .s, all remains same (maximum verbose output)

stevstrong, 2020-06-02:
Please use www.stm32duino.com for such issues.

78edu, 2020-06-02:
> 
> 
> Please use [www.stm32duino.com](http://www.stm32duino.com) for such issues.

can you explain details? it won't compile with stm32duino with manually added library.
should i use cmsis dsp instead of stm_fft?
I prefer Arduino STM32 more than stm32duino, it works much faster and i need stm_fft to work.

`c:/users/1/appdata/local/arduino15/packages/stm32/tools/arm-none-eabi-gcc/8.2.1-1.7/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\1\AppData\Local\Temp\arduino_build_326228\sketch\sketch_may17a.ino.cpp.o: in function `loop':

sketch_may17a.ino.cpp:(.text.loop+0x26): undefined reference to `cr4_fft_1024_stm32'

collect2.exe: error: ld returned 1 exit status`

stevstrong, 2020-06-03:
I suggested to use the **forum** www.stm32duino.com

rogerclarkmelbourne, 2020-06-10:
Why 10.07.2020

stevstrong, 2020-06-11:
I suggest to change the extension "*.s" to "*.S" (capital S), then it should compile.

stevstrong, 2020-10-12:
Closed due to inactivity.

closed 2020-10-12 by stevstrong