rogerclarkmelbourne/Arduino_STM32 #801: micadc example: no matching function for call to 'STM32ADC::setDMA(uint16_t [2], int, int, NULL)'

jreyesr, 2020-06-16:
There is a compilation error in the micadc example for the USBComposite library. It fails with an error `no matching function for call to 'STM32ADC::setDMA(uint16_t [2], int, int, NULL)'`.
AFAIK, it's the line
https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/9b81135699acc7356721957d7013fde052a0b55f/STM32F1/libraries/USBComposite/examples/micadc/micadc.ino#L29
The parameters for `setDMA` have changed and that invocation is no longer valid. I'm preparing a pull request that (should) solve it.

jreyesr, 2020-06-16:
Update: Two of the STM32ADC examples are also broken.
I think https://github.com/rogerclarkmelbourne/Arduino_STM32/issues/769#issuecomment-642701526 refers to this bug, the attached file implements the same fix.

stevstrong, 2020-06-17:
I think 5236648 fixes the issue without changing all those examples.

commit 5236648a (2020-06-17): F1: update STM32ADC.h to fix #801
closed 2020-06-17 by stevstrong